Phylliade/ikpy #9: Importing two Chain objects from URDF raises an unclear error

s-mick, 2016-02-24:
I tried to create two distinct `Chain` objects, both of them being imported from URDF files:

```python
chain1 = Chain.from_urdf_file("poppy_ergo.URDF")
chain2 = Chain.from_urdf_file("poppy_ergo.URDF")
```

The first object is OK and works well as long as no other URDF import is performed. I can include several kinematics calculations between these two instanciations, and nothing wrong happens. But when instanciating the second object, I get an `AttributeError`:

`
AttributeError: 'NoneType' object has no attribute 'find'
`

coming from

```python
.../site-packages/ikpy/URDF_utils.py in find_next_link
    53    if next_link_name is None:
    54        # If the name of the next link is not provided, find it
--> 55        next_link_name = current_joint.find("child").attrib["link"]
```

This error occurs:

* only when URDF import is used for the second time: multiple manually-built chains seem to work fine
* whether the involved URDF files are different or not
* whether the import creates a new `Chain`object or overwrites an old one

Phylliade, 2016-02-24:
Ok, i think it may be an error in the opening of the XML file. (The XML object may be reused, instead of reinstantiated). I will check this out!

Phylliade, 2016-03-26:
Fixed in https://github.com/Phylliade/ikpy/commit/6882b460b6eaeb86e13e24c5f8651f05bb5196d6

closed 2016-03-26 by Phylliade