Phylliade/ikpy #10: Requirements.txt is not consisten with setup.py

show0k, 2016-03-15:
```python
In [1]: import ikpy
ImportError                               Traceback (most recent call last)
<ipython-input-1-7df5a6184463> in <module>()
----> 1 import ikpy

/Volumes/data/home/miniconda2/lib/python2.7/site-packages/ikpy/__init__.py in <module>()
----> 1 from . import chain

/Volumes/data/home/miniconda2/lib/python2.7/site-packages/ikpy/chain.py in <module>()
      7 from . import URDF_utils
      8 from . import inverse_kinematics as ik
----> 9 from . import plot_utils
     10 import numpy as np
     11 from . import link as link_lib

/Volumes/data/home/miniconda2/lib/python2.7/site-packages/ikpy/plot_utils.py in <module>()
      1 # coding= utf8
----> 2 import matplotlib.pyplot
      3 import numpy as np
      4 from . import geometry_utils
      5 import matplotlib.animation

ImportError: No module named matplotlib.pyplot
```
You should load your requirements.txt in your setup.py, or refactor your code to make matplotlib an optional dependency.

Phylliade, 2016-08-08:
Hi Theo!
This should have been fixed in https://github.com/Phylliade/ikpy/commit/fe8c07b69f587c6825d51dd0a66183345642dadf.

show0k, 2016-09-01:
Great, thank you !

closed 2016-09-01 by show0k