Phylliade/ikpy #12: error trying to import forward_kinematics

studywolf, 2016-06-15:
Hello! I was trying to run the Quickstart tutorial for poppy-humanoid, and it's throwing an import error looking for the forward_kinematics file, which (after some digging) i found was deleted in this commit 164a5ba680191172f0177e9c59edc5939ce1d114 

Should it still be around, or should that import be removed? It looked like it wasn't just a stray import because fk functions are used throughout the code, should the forward kinematics file be re-added or should all the references be removed? 

thanks!

Phylliade, 2016-06-15:
Hi!
The `forward_kinematics` module has been merged directly in the `link` and `chain` modules, so it shouldn't be used anymore.
About your error, do you know which notebook was raising this import error?

Thanks for your feedback!

studywolf, 2016-06-15:
Hi! great thanks! 

it's in this tutorial https://github.com/Phylliade/ikpy/blob/master/tutorials/poppy-inverse-kinematics/Quickstart.ipynb

when it was importing model the error gets thrown.
thanks!

Phylliade, 2016-06-15:
Hi again,
Actually, the `poppy-inverse-kinematics` module (and the tutorials) is deprecated. I should remove it. 
A working quickstart could be found at : https://github.com/Phylliade/ikpy/blob/master/tutorials/ikpy/Quickstart.ipynb
This one should work smoothly. (But i would be happy to fix if something is wrong!).

studywolf, 2016-06-15:
Oh great, thanks!

closed 2016-06-15 by studywolf

studywolf, 2016-06-15:
oh wait, i grokked what you're saying, poppy is no longer supported?

Phylliade, 2016-06-15:
The poppy ecosystem will remain very close to the library, so it's supported! 
`poppy-inverse-kinematics` is just a preliminary version of the IKPy library. (When changing the codebqse, we also changed the name):

BTW, every Poppy creature is supported, and many other robots, as long as they have an URDF.

studywolf, 2016-06-15:
ah! great! thanks :+1: