Phylliade/ikpy #15: AttributeError: 'Robot' object has no attribute 'goto_target'

studywolf, 2016-06-16:
I'm in the Inverse kinematics with the ErgoJr tutorial, and in cell 7 i'm getting 

```
----
AttributeError                            Traceback (most recent call last)
<ipython-input-7-d1db5050b446> in <module>()
----> 1 poppy.goto([-0.00670386, -0.17439168,  0.15083097])

AttributeError: 'Robot' object has no attribute 'goto'
```

I've tried to track this down but without luck, can't find where the Robot object is stored, any ideas?

Phylliade, 2016-06-22:
I'm on it, i'll check as soon as i have some time!
Did you manage to get the other notebooks (like this [one](https://github.com/Phylliade/ikpy/blob/master/tutorials/ikpy/Quickstart.ipynb)) to work?

Thanks again for your feedback!

studywolf, 2016-06-22:
great thanks! and yup that  other one works perfect!

Phylliade, 2016-06-25:
This is now fixed : the object to use for Inverse Kinematics is actually `poppy.chain`, (not `poppy` itself).

studywolf, 2016-06-25:
awesome, thanks!

Phylliade, 2016-09-13:
Hi again Studywolf,
Is this issue closed? (Just checking which issues require some work!)

studywolf, 2016-09-15:
oh, yes! thanks!

closed 2016-09-15 by studywolf