Phylliade/ikpy #20: inverse, forward kinematics problem

HaijieLi, 2016-07-15:
Hi Phylliade
I am using your code to solve the ur5 kinematics. 
However when I use the conmands inverse_kinematics and forward_kinematics,
the resuts are very strange.
For example in your "Getting Stated"
I simply copped the code like this:

    import numpy as np
    import matplotlib.pyplot as plt
    import ikpy as ik
    

    from ikpy import chain
    from ikpy import plot_utils
    
    my_chain = ik.chain.Chain.from_urdf_file("ik/poppy_ergo.URDF")

    
    
    fk = my_chain.forward_kinematics([  0.00000000e+00,  -7.83311388e-01,  -1.13929769e+00,
         8.39373996e-01,   6.05357632e-05,   7.31474063e-01,
         0.00000000e+00])
                                      
    print(fk)

and it returns

[[ -7.05670648e-01   6.43585169e-01   2.96356318e-01   1.49993875e-01]
 [  7.08540003e-01   6.40968301e-01   2.95179100e-01   1.50195323e-01]
 [  1.78851283e-05   4.18279534e-01  -9.08318354e-01   1.67503064e-01]
 [  0.00000000e+00   0.00000000e+00   0.00000000e+00   1.00000000e+00]]

Not as expected

    [[1, 0, 0, 2],
    [0, 1, 0, 2],
    [0, 0, 1, 2],
    [0, 0, 0, 1]]

Moreover, when I input the transfer matrix fk as above, into a inverse_kinematics, the result is not the same with what I used 

    ik = my_chain.inverse_kinematics(fk)
    
    print(ik)

It returns  
[ 0.         -0.78674925 -1.14065448  0.84229575 -0.00591668  0.72914705
  0.        ]

Am I using the code the right way, or something is wrong.

Thanks a lot!

Phylliade, 2016-08-08:
Hi HaijieLi,
This was indeed an incorrect value from another version of the robot, this is now fixed on the [guide](https://github.com/Phylliade/ikpy/wiki)!
You can also try the live version in the [notebook](https://github.com/Phylliade/ikpy/blob/master/tutorials/ikpy/Quickstart.ipynb)

Thanks for your feedback!

HaijieLi, 2016-08-11:
Thanks for your help!

Phylliade, 2016-09-13:
Hi HaijeLi,
Since this is issue is open, is there something not working in the library?

closed 2019-06-09 by Phylliade