Phylliade/ikpy #21: Creating chain from URDF

mjm522, 2016-07-18:
Hi, I am trying to create a chain from URDF file of Baxter and is facing following error. I can't figure out why? What is wrong in this declaration? Any help please
```
my_chain = ikpy.chain.Chain.from_urdf_file("../baxter_common/baxter_description/urdf/baxter.urdf",
    base_elements=["right_upper_shoulder", "right_lower_shoulder", "right_upper_elbow", "right_lower_elbow", 
                   "right_upper_forearm", "right_lower_forearm", "right_wrist", "right_hand"], 
    last_link_vector=[0.0, 0.0, 0.025], 
    active_links_mask=[False, True, True, True, True, True, True,True])
```

I am using this file: [Baxter URDF](https://github.com/RethinkRobotics/baxter_common/blob/master/baxter_description/urdf/baxter.urdf)

And get this error!!!

```
raise ValueError("Your active links mask length of {} is different from the number of your links, which is {}".format(len(active_links_mask), len(self.links)))
ValueError: Your active links mask length of 8 is different from the number of your links, which is 2
```

Phylliade, 2016-08-08:
Hi mjm522, I'll have a look and come back to you as soon as possible!

1Faith, 2016-08-31:
@mjm522  I have the same problem with the Pirobot. Did you already find a solution?

I defined my base_elements as a list of strings that includes all the links of between the first link of Pirobot ("base_footprint") and the last link of its arm ("right_gripper_link"). My chain is then defined as:

`my_chain = ikpy.chain.Chain.from_urdf_file(".../rbx2/rbx2_description/urdf/pi_robot/urdf_pi_robot_with_arm_0.urdf", base_elements=["base_footprint", "base_link", "torso_link", "right_arm_base_link","right_arm_shoulder_F2_0_link","right_arm_shoulder_pan_link", "right_arm_shoulder_F3_0_link", "right_arm_shoulder_lift_link", "right_arm_shoulder_F3_1_link", "right_arm_shoulder_roll_link", "right_arm_shoulder_roll_F3_0_link", "right_arm_shoulder_roll_F9_0_link", "right_arm_elbow_F3_0_link", "right_arm_elbow_flex_servo_link", "right_arm_elbow_flex_link", "right_arm_forearm_F3_0_link", "right_arm_forearm_flex_servo_link", "right_arm_forearm_flex_link", "right_arm_forearm_F2_0_link", "right_arm_wrist_flex_link", "right_arm_wrist_F3_0_link", "right_arm_gripper_attach_link", "right_gripper_base_link", "right_gripper_link"], active_links_mask=[False, False, False, False, True, True, True, True, True, True, True, True, True, True, True, True, True, True, True, True, True, True, True], base_element_type="link")`

This results in the following error:

`File .../.local/lib/python2.7/site-packages/ikpy/chain.py", line 33, in __init__
    raise ValueError("Your active links mask length of {} is different from the number of your links, which is {}".format(len(active_links_mask), len(self.links)))
ValueError: Your active links mask length of 23 is different from the number of your links, which is 1
terminate called after throwing an instance of 'boost::exception_detail::clone_impl<boost::exception_detail::error_info_injector<boost::lock_error> >'
  what():  boost: mutex lock failed in pthread_mutex_lock: Invalid argument
Aborted (core dumped)`

Phylliade, 2016-09-13:
Hi mjm522 and 1Faith, 
Actually, the importer is creating a blank chain from the Baxter robot. (That's exactly the same for the Pirobot error).
Maybe it's caused by the slight deviations in the URDF format : i will investigate that. 
@1Faith, could you provide the URDF file for the Pirobot?

1Faith, 2016-09-14:
@Phylliade Thanks for your reply.

My URDF file is made from an adapted xacro-file based on: [https://github.com/pirobot/rbx2/blob/indigo-devel/rbx2_description/urdf/pi_robot/pi_robot_with_arm.xacro](url) Underlying files were also adapted, basically to remove the wheels and functionality that enables it to move in a plane, because it was causing problems.

I hope you can identify the problem.

[urdf_pi_robot_with_arm_0.urdf.tar.gz](https://github.com/Phylliade/ikpy/files/471751/urdf_pi_robot_with_arm_0.urdf.tar.gz)

Rogerpi, 2017-02-09:
Check https://github.com/Phylliade/ikpy/blob/master/src/ikpy/chain.py , from_urdf_file function

Maybe the issue is that you have a diferent base link name. I used a urdf file generated from a xacro file and I had the same trouble. Change it in the arguments:

my_chain = ikpy.chain.Chain.from_urdf_file("./src/moveo/urdf/moveo.urdf",base_elements=["base"])

"./src/moveo/urdf/moveo.urdf" is my urdf file
"base" the base link name on the urdf file

JulianYG, 2017-07-06:
I'm having the same issue on sawyer/kuka -- it's not reading in blank contents. I couldn't find any inconsistency in URDF format.

JulianYG, 2017-07-07:
I noticed I just need to change the base elements. However, the z axis value is very off.

closed 2020-05-10 by stale[bot]