Phylliade/ikpy #24: Le lancement du fichier poppy_torso.URDF

JingleiSHI, 2016-08-01:
J'ai eu un problème quand je lance les codes:  dans le dossier 'src', la classe chain a une fonction 'from_urdf_file' qui peut lire tous les joints dans le fichier URDF, mais quand je utilise le fichier 'poppy_torso.URDF' dans le dossier 'source', la fonction 'from_urdf_file' ne peut pas lire les joints, mais avec le fichier 'poppy_torso.URDF', cela marche.
Est-ce que cette faute est a cause de ma mal compréhension des codes ou je manque quelques choses ?
En plus, monsieur, si cela ne vous déranger pas, est-ce que vous pourriez m'expliquer comment faire la transformation entre les angles des joints et leurs positions Cartisienne, maintenant, j'ai une liste d'angle de chaque joint, avec la fonction 'forward_kinematics' , je peux avoir la matrice f-k, et puis, qu'est-ce que je doit faire?

Phylliade, 2016-09-13:
Hi KingleSHI, 
First, if you don't mind, can we continue in English? (To be sure to be understood by everybody).
I'm not sure to understand the first issue, is this working? :
```
my_chain = ikpy.chain.Chain.from_urdf_file("resources/poppy_ergo.URDF")
```

For the second issue, if you only need Forward Kinematics (FK), you're ready to go! You can also plot it. (More details in the links below).
If you want to do inverse Kinematics, you can try this tutorial, to have a complete list of steps to follow:
https://github.com/Phylliade/ikpy/blob/master/tutorials/ikpy/Quickstart.ipynb
(or a non notebook version : https://github.com/Phylliade/ikpy/wiki)

Phylliade, 2016-12-01:
Hi KingleiSHI, is the issue closed?

closed 2020-05-10 by stale[bot]