Phylliade/ikpy #26: Solving not only the position but orientation including

franke1276, 2016-08-25:
I try to solve IK for my robot. Is there any way to solve not only the position but the orientation, too?

I need a IK-Solver with such function to get the angles for my joints depending on the orientation for the end-effector of my robot. Is that possible with your library?

Phylliade, 2016-09-13:
Hi franke1276,
This feature is not implemented yet, but, if you want, you can add it by using the [dev API](https://github.com/Phylliade/ikpy/wiki/Developper-API), it should be pretty straightforward.

show0k, 2016-10-13:
Hi, 
I have absolutely no idea what it worth, but someone on the poppy forum [apparently manage ](https://forum.poppy-project.org/t/inverse-kinematic-with-pen-holder-poppy-ergo-jr/2601/7) to add a orientation constraint to the end effector for drawing with a Poppy Ergo Jr. 
It basically change the optimization function from L-BFGS-B to SLSQP which enable to add an angle constraints on the ouput.

AhmetMericOzcan, 2019-08-20:
I believe giving constraints are not enough for orientation.

MrDadaGuy, 2019-08-22:
Also need this functionality - did anyone find a solution for this?

Phylliade, 2020-05-02:
Orientation has landed in master!

This was added in https://github.com/Phylliade/ikpy/pull/79

closed 2020-05-02 by Phylliade