Phylliade/ikpy #30: Unsing standalone ros URDF parser instead of yours

show0k, 2017-01-19:
One year ago ROS people published a [standalone python URDF parser ](https://github.com/ros/urdf_parser_py). It could be interesting to use this one to be sure that every URDF supported by ROS works with ikpy.
It could avoid issues like #21.

Phylliade, 2017-03-16:
Hi @show0k, 
Yes, it seems a good idea! Do you want to do it, or is it more an idea of an enhancement for now?

show0k, 2017-03-21:
It was more an idea of enhancement, to keep the link of the URDF python parser of ROS.

Phylliade, 2020-05-03:
Hello @show0k, pinning this as it's still something to try

Phylliade, 2026-10-26:
Revisiting this after the 4.2.0 release. Switching to `urdf_parser_py` would mean a new dependency and a rewrite of the chain extraction, and the compatibility gaps it was meant to close have been fixed one by one in IKPy's own parser: `continuous` joints, the default joint axis, the `base_elements` lookup errors, fixed joints, mimic-free chains from any link. No open issue depends on it anymore.

Closing as not planned. If a URDF that loads in ROS fails in IKPy, please open a dedicated issue with the file; that is the useful signal, and it gets fixed in the parser directly.

closed 2026-10-26 by Phylliade