Phylliade/ikpy #35: How to create chain of links from DH table, I got an error while trying to do that.

Abhi001vj, 2017-08-11:
`l0 = ikpy.link.OriginLink()
l1 = ikpy.link.DHLink('l1', d=3, a=0, bounds=None, use_symbolic_matrix=True)
l2 = ikpy.link.DHLink('l2', d=0, a=5.75, bounds=None, use_symbolic_matrix=True)
l3 = ikpy.link.DHLink('l3', d=0, a=7.375, bounds=None, use_symbolic_matrix=True)
l4 = ikpy.link.DHLink('l4', d=0, a=0, bounds=None, use_symbolic_matrix=True)
l5 = ikpy.link.DHLink('l5', d=4.125, a=0, bounds=None, use_symbolic_matrix=True)
my_chain = ikpy.chain.Chain([l0,l1,l2,l3,l4,l5])
`
I got this error 
`---------------------------------------------------------------------------
AttributeError                            Traceback (most recent call last)
<ipython-input-13-38ae36d9eb71> in <module>()
----> 1 my_chain = ikpy.chain.Chain([l0,l1,l2,l3,l4,l5])

~/.local/lib/python3.5/site-packages/ikpy/chain.py in __init__(self, links, active_links_mask, name, profile, **kwargs)
     21         self.name = name
     22         self.links = links
---> 23         self._length = sum([link._length for link in links])
     24         # Avoid length of zero in a link
     25         for (index, link) in enumerate(self.links):

~/.local/lib/python3.5/site-packages/ikpy/chain.py in <listcomp>(.0)
     21         self.name = name
     22         self.links = links
---> 23         self._length = sum([link._length for link in links])
     24         # Avoid length of zero in a link
     25         for (index, link) in enumerate(self.links):

AttributeError: 'DHLink' object has no attribute '_length'`

Also I got this error when trying to import  from ikpy import plot_utils

`from ikpy import plot_utils`

error 

`---------------------------------------------------------------------------
ImportError                               Traceback (most recent call last)
<ipython-input-14-c685a15408be> in <module>()
      1 import ikpy
      2 import numpy as np
----> 3 from ikpy import plot_utils

~/.local/lib/python3.5/site-packages/ikpy/plot_utils.py in <module>()
      4 from . import geometry_utils
      5 import matplotlib.animation
----> 6 from mpl_toolkits.mplot3d import Axes3D
      7 
      8 

/usr/lib/python3/dist-packages/mpl_toolkits/mplot3d/__init__.py in <module>()
      2                         unicode_literals)
      3 
----> 4 from matplotlib.externals import six
      5 
      6 from .axes3d import Axes3D

ImportError: No module named 'matplotlib.externals'`

Thank you.

ChunJyeBehBeh, 2019-09-14:
Hello, may I know have you solved the problem?

Phylliade, 2019-10-11:
Hey,

This seems to be a bug in the DH links.

Actually the DH implementation is preliminary and lacks uses-cases: is the code you presented coming from a real robot? In this case, could you provide it so I can use it as a base for the test cases to develop further the DH parameters?

Phylliade, 2019-10-11:
Additionally, the second issue you got (with matplotlib) is an ubuntu issue.

Take a look at this, this may help you:
https://stackoverflow.com/questions/45103248/importerror-no-module-named-matplotlib-externals

Tanneguydv, 2023-07-04:
I written some code to resolve the problem of creating a chain link from DH parameters table, see https://github.com/Phylliade/ikpy/pull/143

Phylliade, 2024-07-28:
DH are now included in the last release!

closed 2024-07-28 by Phylliade