Phylliade/ikpy #37: How to create a Chain without an URDF file?

marcosvilela, 2017-09-15:
I'm aiding a 3 DOF manipulator development, and I'm looking through the library to use it on the project. But, as I'm helping making it from scratch, there is no URDF file to use, and the examples on the tutorial pages are always using it. How can I code a chain without having it previously on an URDF file?

frueter, 2018-03-06:
Same here. I tried to look up how to manually create a Chain but the docs page for the Link object is blank.
A simple example would be much appreciated.

maximkulkin, 2018-06-13:
Guys, I have updated [Homepage](https://github.com/Phylliade/ikpy/wiki) with instructions on how to create chains manually.

vivekpatel99, 2018-11-27:
@maximkulkin can you please provide me with some information, How can I get those parameters of "chain" ("Creating a chain manually") for the robotic arm?

Phylliade, 2020-05-03:
Hey everyone, 

I'm currently working on streamlining the workflow to create new robots.
In the last version, new ready-to-use chains were added, in the form of JSON files that add useful metadata to a URDF file. It just works as below:

```
my_chain = Chain.from_json_file("baxter_left_arm.json")

Regarding the original issue, i'm going to add a simple tutorial on how to create a chain from scratch, in the next release

Phylliade, 2020-05-03:
@marcosvilela, would you have any description of the chain you are currently (actually were ;) ) using?

Phylliade, 2026-10-26:
Creating a chain without a URDF file is supported and documented:

- Build it from links directly: `Chain([OriginLink(), URDFLink(name=..., origin_translation=..., origin_orientation=..., rotation=...), ...])`. The `URDFLink` and `DHLink` parameters are documented at https://ikpy.readthedocs.io/en/latest/link.html, and the wiki page "Creating a chain manually" walks through an example.
- From a Denavit-Hartenberg table with `DHLink` (see `tests/test_chain_dh.py` for a UR10 example). The next release also accepts `convention="modified"`.

Closing, as the original question is answered. Open a new issue if a specific parameter is unclear.

closed 2026-10-26 by Phylliade