Phylliade/ikpy #38: Jacobian Implementation

mikekaram, 2017-09-19:
Hello Phylliade,

I am trying to use your code for a 6-dof manipulator I am building. However I want to take the Jacobian from my created chain and there isn't such function. Will you implement it in the near future? After I saw that, I am trying to implement it myself, but I am having trouble as I want the symbolic transformation matrix of each link, then multiply them all and get the final (symbolic) transformation matrix. After that it's a matter of differentiation with respect to the thetas I plug in. However the symbolic_transformation_matrix of every link is a function and not a sympy matrix, so i can't multiply them all together. Apart from that I want to specify the names of the variables (of the "thetas") for every symbolic matrix(every variable is called "theta"). Could you please help me in any way?

Phylliade, 2020-05-03:
Hello @mikekaram, sorry for the late reply.

Support for a custom Jacobian is currently not possible with the current implementation. 
I'm thinking about supporting alternative optimization schemes

Phylliade, 2026-10-26:
This is now available through the JAX backend (IKPy 4.1.0, `pip install 'ikpy[jax]'`): the kinematics are differentiable, and the inverse kinematics uses the analytical Jacobian of the residual computed by `jax.jacfwd`. The symbolic (SymPy) path is kept only for the per-link matrices, and is not the way to get a Jacobian anymore. If you need the Jacobian of the end-effector pose itself for your own solver, `jax.jacfwd` over `ikpy.jax_backend.forward_kinematics_jax` gives it directly.

Closing; a dedicated `Chain.jacobian` helper can be a separate feature request if there is demand.

closed 2026-10-26 by Phylliade