Phylliade/ikpy #39: Limits issue

Fenrir12, 2017-10-17:
Hi,

I've seen this issue brought up in other issues and I'm having it as well. 
My urdf is fairly simple, you can have a look at it:
<robot name="arm_v0">
	<link name="base_link">
		<visual>
			<geometry>
        			<box size="0.3 0.25 0.1"/>
        		</geometry>
		</visual> 
	</link>
	<link name="platform_link">
		<visual>
			<geometry>
        			<cylinder length="0.01" radius="0.1"/>
        		</geometry>
		</visual> 
	</link>
	<link name="arm_link"> 
		<visual>
			<geometry>
        			<box size="0.215 0.2 0.2"/>
        		</geometry>
		</visual> 
	</link>
	<link name="forearm_link"> 
	</link>
	<link name="gripper_link"> 
	</link>

	<joint name="rot_joint" type="revolute">
    		<parent link="base_link"/>
    		<child link="platform_link"/>
    		<origin xyz="0 0 0.09" rpy="0 0 0" /> 
    		<axis xyz="0 0 1" />  
  	</joint>

  	<joint name="shoulder_joint" type="revolute">
    		<parent link="platform_link"/>
    		<child link="arm_link"/>
    		<origin xyz="0 0 0.05" rpy="0 0 0" /> 
    		<axis xyz="0 1 0" /> 
		<limit effort="3.1" lower="-0.02" upper="0.02" velocity="7.0"></limit>
 	</joint>

  	<joint name="elbow_joint" type="revolute">
    		<parent link="arm_link"/>
    		<child link="forearm_link"/>
    		<origin xyz="-0.037 0 0.172" rpy="0 0 0" />
		<limit effort="3.1" lower="-0.2" upper="0.2" velocity="7.0"/> 
    		<axis xyz="0 1 0" /> 
  	</joint>

  	<joint name="grip_joint" type="revolute">
    		<parent link="forearm_link"/>
    		<child link="gripper_link"/>
    		<origin xyz="0.135 0 -0.172" rpy="0 0 0" /> 
    		<axis xyz="0 1 0" /> 
		<limit effort="3.1" lower="-0.2" upper="0.2" velocity="7.0"/> 
  	</joint>
</robot>
 
In the shoulder joint, I've set the limit between -0.2 and 0.2 rad, however it doe snot respect the limit at all, am I missing something?

Thanks,

Samuel

FOXNEOAdvancedTechnology, 2017-11-08:
There is a way to specify bounds as a tuple in a chain, but it doesn't appear to be read in from the URDF.

romarcg, 2018-05-12:
I was having the same issue. It seems that the parameter from the *limits* element (in the URDF) is not recovered. From file URDF_utils.py:

```
def get_urdf_parameters(urdf_file, base_elements=["base_link"], last_link_vector=None, base_element_type="link"):
    """Returns translated parameters from the given URDF file
   :param urdf_file: The path of the URDF file
   :type urdf_file: string
   :param base_elements: List of the links beginning the chain
   :type base_elements: list of strings
   :param last_link_vector: Optional : The translation vector of the tip.
   :type last_link_vector: numpy.array
    """
   .
   .
   .

    # Save the joints in the good format
    for joint in joints:
        translation = joint.find("origin").attrib["xyz"].split()
        orientation = joint.find("origin").attrib["rpy"].split()
        rotation = joint.find("axis").attrib['xyz'].split()
        parameters.append(lib_link.URDFLink(
            translation_vector=[float(translation[0]), float(translation[1]), float(translation[2])],
            orientation=[float(orientation[0]), float(orientation[1]), float(orientation[2])],
            rotation=[float(rotation[0]), float(rotation[1]), float(rotation[2])],
            name=joint.attrib["name"]
        ))

    # Add last_link_vector to parameters
    if last_link_vector is not None:
        parameters.append(lib_link.URDFLink(
            translation_vector=last_link_vector,
            orientation=[0, 0, 0],
            rotation=[0, 0, 0],
            name="last_joint"
        ))

    return(parameters)

```
The fix I choose was to rewrite this function and call it instead of the original one (I am using a packaged version of this lib). After doing this IK results seem more coherent (not jumping out of bounds).

Add the *bound* lines to the aforementioned function:

```
# Save the joints in the good format
    for joint in joints:
        translation = joint.find("origin").attrib["xyz"].split()
        orientation = joint.find("origin").attrib["rpy"].split()
        rotation = joint.find("axis").attrib['xyz'].split()
        bounds = [joint.find("limit").attrib['lower'],joint.find("limit").attrib['upper']]
        parameters.append(lib_links.URDFLink(
            translation_vector=[float(translation[0]), float(translation[1]), float(translation[2])],
            orientation=[float(orientation[0]), float(orientation[1]), float(orientation[2])],
            rotation=[float(rotation[0]), float(rotation[1]), float(rotation[2])],
            bounds=[float(bounds[0]), float(bounds[1])],
            name=joint.attrib["name"]
        ))
```

Phylliade, 2019-03-17:
Bounds are now taken into account thanks to #42

closed 2019-03-17 by Phylliade