Phylliade/ikpy #44: AttributeError: 'DHLink' object has no attribute '_length'

C-Aniruddh, 2018-07-14:
The above error arises because the ```self._length``` parameter is not defined in the DHLink class (found at https://github.com/Phylliade/ikpy/blob/master/src/ikpy/link.py)

So, if I try to initialize the bot by using the following code : 

```
# Define the bot based on Denavit-Hartenberg values
robot_chain = Chain(name="robotic_chain", links=[
    DHLink(
        name="base",
        d=65,
        a=0
    ),
    DHLink(
        name="shoulder",
        d=0,
        a=241
    ),
    DHLink(
        name="elbow",
        d=0,
        a=165
    ),
    DHLink(
        name="wrist_pitch",
        d=0,
        a=110
    ),
    DHLink(
        name="wrist_roll",
        d=110,
        a=0
    )
])
```
I would end up with an error : "```AttributeError: 'DHLink' object has no attribute '_length'```". Does anyone know how to calculate the ```_length``` from the DH Parameters?

susanto-tm, 2019-02-19:
@C-Aniruddh I am experiencing the same issue at the moment. Have you found a way to solve this issue?

3dprintingdoctor, 2019-05-21:
I am experiencing the same issue at the moment. Have you found a way to solve this issue?

closed 2020-05-10 by stale[bot]