Phylliade/ikpy #46: ik angles for more than one axis of rotation

svabhishek29, 2018-08-30:
Hi,

i was trying out the sample code and modified the chain to as below 
`left_arm_chain = Chain(name='left_arm', links=[
    OriginLink(),
    URDFLink(
        name="shoulder",
        translation_vector=[1, 0, 0],
        orientation=[0, 0, 0],
        rotation=[0, 1, 1],
    ),
    URDFLink(
        name="elbow",
        translation_vector=[10, 0, 0],
        orientation=[0, 0, 0],
        rotation=[0, 0, 1],
    ),
    URDFLink(
        name="wrist",
        translation_vector=[10, 0, 0],
        orientation=[0, 0, 0],
        rotation=[0, 0, 1],
    )
])`

the ik plot is showing properly.

but i am not able to understand how the angles are being printed.

when i run 
`print(left_arm_chain.inverse_kinematics([[1, 0, 0, 11],
                                         [0, 1, 0, 6],
                                         [0, 0, 1, 6],
                                         [0, 0, 0, 1]]))`

i get the output 
`[ 0.         -0.6926806   2.01853327  0.        ]`

shouldn't the first rotation angle have 2 angles? one in z and another in y?

please help me understand how to get 2 angles from this.

Thanks

Phylliade, 2020-05-03:
Hello @svabhishek29, sorry for the late reply.

Each link corresponds to one rotation axis.
In your first Link, you don't specify two rotation axes, but one rotation axis with the coordinates [0, 1, 1]

If you want to have two rotations, you should specify two links

Phylliade, 2020-05-03:
Closing this for now, don't hesitate to reopen the issue if needed

closed 2020-05-03 by Phylliade