Phylliade/ikpy #56: Create Chain method will not make chain from URDF file and also does not throw an error

Lutarp, 2019-06-04:
Hi, while using the library on a raspberry PI, the .from_urdf_file() method seems to run fine on the tutorial file poppy_ergo.URDF. But when given my URDF file to create a chain it does not actually create a chain, merely creating a simple base_link fill in. It also does not throw any error when trying to parse the file, it simply takes it and does not parse any data.
[image]
[image]

Phylliade, 2019-06-09:
Hey @Lutarp, could you paste your URDF file?

closed 2020-05-10 by stale[bot]

jurka14, 2021-03-03:
Hi, I have been having the same problem with my URDF file. Here it is:

`
<?xml version="1.0" encoding="utf-8"?>

<robot
  name="Rameno_sestava.SLDASM">
  <link
    name="base_Link">
    <inertial>
      <origin
        xyz="0.0375 0.00734688456192442 1.24826504699513E-09"
        rpy="0 0 0" />
      <mass
        value="0.010277230998552" />
      <inertia
        ixx="1.092531894428E-05"
        ixy="1.4467160407856E-21"
        ixz="-2.20849246804654E-13"
        iyy="1.30777709748279E-05"
        iyz="1.98533463602263E-13"
        izz="3.52427256535382E-06" />
    </inertial>
    <visual>
      <origin
        xyz="0 0 0"
        rpy="0 0 0" />
      <geometry>
        <mesh
          filename="package://Rameno_sestava.SLDASM/meshes/base_Link.STL" />
      </geometry>
      <material
        name="">
        <color
          rgba="0.627450980392157 0.627450980392157 0.627450980392157 1" />
      </material>
    </visual>
    <collision>
      <origin
        xyz="0 0 0"
        rpy="0 0 0" />
      <geometry>
        <mesh
          filename="package://Rameno_sestava.SLDASM/meshes/base_Link.STL" />
      </geometry>
    </collision>
  </link>
  <link
    name="Rotating_base_Link">
    <inertial>
      <origin
        xyz="-0.0310067493964589 0.0356912345580542 -1.40021952366176E-11"
        rpy="0 0 0" />
      <mass
        value="0.0842744537549328" />
      <inertia
        ixx="3.72775016732398E-05"
        ixy="9.39829888441467E-06"
        ixz="3.46042252569305E-10"
        iyy="3.97404908918724E-05"
        iyz="2.56760938255436E-10"
        izz="3.12686222723885E-05" />
    </inertial>
    <visual>
      <origin
        xyz="0 0 0"
        rpy="0 0 0" />
      <geometry>
        <mesh
          filename="package://Rameno_sestava.SLDASM/meshes/Rotating_base_Link.STL" />
[... 244 lines cut ...]
        <mesh
          filename="package://Rameno_sestava.SLDASM/meshes/End_point_Link.STL" />
      </geometry>
    </collision>
  </link>
  <joint
    name="End_point_Joint"
    type="revolute">
    <origin
      xyz="0 -0.003 -0.12435"
      rpy="-2.1581 0 -3.1416" />
    <parent
      link="Barrel_part_Link" />
    <child
      link="End_point_Link" />
    <axis
      xyz="1 0 0" />
    <limit
      lower="-0.98347"
      upper="2.1581"
      effort="0"
      velocity="0" />
  </joint>
</robot>
`