Phylliade/ikpy #64: Numba/CUDA compatible?

blooop, 2019-10-02:
Hi,

I'm interested in running say 1000 independent IK queries on a gpu. In the readme you mention that this is a pure python library which means it theory it should work with numba relatively easily.  Are there any caveats or thing I should be aware of if I'm going to try this?

Something similar to this:
https://ipython-books.github.io/58-writing-massively-parallel-code-for-nvidia-graphics-cards-gpus-with-cuda/

Thanks

Phylliade, 2019-10-11:
Hey @blooop,

That's a good idea indeed! 
There is no caveat a priori for the inference part (forward kinematics)
For the inverse kinematics, the optimization is done in scipy, out of the python loop, so it won't be done on the GPU.

FYI, there is a plan to use Jax (https://github.com/google/jax) to do the inverse kinematics on GPU

blooop, 2019-10-11:
Thanks, jax also looks very interesting.

benjirer, 2024-09-16:
Hey! It's been a while but is there an update on this? Would be super useful!

Phylliade, 2024-09-18:
Let me put up on the priority list then!

Phylliade, 2026-10-26:
This is now covered by the JAX backend introduced in IKPy 4.1.0: `pip install 'ikpy[jax]'` and pass `backend="jax"` to `inverse_kinematics`. Forward kinematics and the IK residual/Jacobian are JIT-compiled and run on whatever device JAX is configured for, including GPU. Batched queries can be vectorized with `jax.vmap` over the compiled kinematics.

See the "JAX Backend" section of the README. Closing.

closed 2026-10-26 by Phylliade