Phylliade/ikpy #66: Ways to deal with self-collisions?

beduffy, 2019-11-07:
Hello, great library, thanks for your work!

Is there any way to ensure that the joint angles returned from inverse kinematics won't result in a self-collision? I have my own functions for detecting these but how could I influence or change the optimisation to avoid them?

Phylliade, 2019-12-04:
Hey @beduffy,

Currently there is no support for avoiding self-collisions.

However it would be interesting to add this feature!

In order to see where we can start, could you past/explain a bit your self-collision avoidance functions?

beduffy, 2019-12-17:
[image]

Hopefully this image makes it clearer, the blue line is going through the other blue line. If I told the real robot arm to do this there would be damage. 

Generally it's quite easy to detect collisions between line segments e.g. don't include the exact endpoints since two lines share joint locations, loop through each line segment against every other line segment and return True if there is a collision.

Phylliade, 2019-12-23:
Got it!

Yep, that would be indeed possible to add a collision detection function, as described.

Additionally, there is also the topic of collision avoidance, aka, proposing an IK solution that avoids collisions (e.g. not only detecting collisions, but also avoid them). Would you have any ideas on how to implement this?

beduffy, 2020-02-11:
I know you use scipy.optimise underneath so this is akin to a Jacobian/Least Squares method but in the past I've been confused over exactly how ikpy is "symbolic" or maybe it uses sympy for something else. For jacobian methods it's much harder avoid collisions because we have to constrain the optimisation away from these regions and nothing is guaranteed. 

Even just avoiding joint limits is a pain, this master's thesis goes deeper into that: https://pdfs.semanticscholar.org/673d/1b06d968216933d298e0eecff9fa20a2514b.pdf

I prefer analytical/geometric methods since they are faster and it's possible to directly get multiple solutions out of them. But if I was to make a jacobians method to work with collisions it quickly turns into a motion planning problem and you would have to look into OMPL, MoveIt and other libraries. Some methods generally try thousands of configurations close to the target and slowly prune out the configurations which are in collision with objects. Trac-ik (https://arxiv.org/pdf/1811.00600.pdf) and relaxedIK are also other resources. Sorry I can't be of more help.

closed 2020-02-11 by beduffy

Phylliade, 2020-02-15:
Thanks for the resources, will dig into that!