Phylliade/ikpy #70: ikpy is not pure python because it depends on numpy and scipy

nevercast, 2020-01-05:
Since numpy and scipy are not pure python (have c / native dependencies), ikpy cannot be considered pure python. (Unless a slower implementation existed that did not depend on numpy or scipy.

Phylliade, 2020-01-05:
Well you're right, there are C dependencies, but IKPy codebase itself is in pure python.

(executing C code is not a problem per se, the Cpython runtime is in C ;) )

nevercast, 2020-01-07:
> (executing C code is not a problem per se, the Cpython runtime is in C ;) )

I was looking for a pure Python IK library that I could run on MicroPython (embedded systems), your library came up but unfortunately MicroPython does not provide scipy, and numpy is very limited at this time. 

I'm not trying to be pedantic, I really did want to use it in a purely Python environment ❤️

Phylliade, 2020-01-10:
Ah ok got it!

I suggest adding a FAQ about dependencies and the supported platforms.

Would it solve the issue?

closed 2020-05-10 by stale[bot]