Phylliade/ikpy #74: ParseError: not well-formed (invalid token): line 45, column 93

Alexa875, 2020-01-15:
**Hello
Sorry but I am not a coder.
I will try to follow your step to step with poppy Urdf in a file and i have this error**

my_chain = ikpy.chain.Chain.from_urdf_file ("C:\POPPY\poppy_ergo.URDF")
Traceback (most recent call last):

  File "C:\Users\alexa\Anaconda3\lib\site-packages\IPython\core\interactiveshell.py", line 3326, in run_code
    exec(code_obj, self.user_global_ns, self.user_ns)

  File "<ipython-input-4-b502ce84c695>", line 1, in <module>
    my_chain = ikpy.chain.Chain.from_urdf_file ("C:\POPPY\poppy_ergo.URDF")

  File "C:\Users\alexa\Anaconda3\lib\site-packages\ikpy\chain.py", line 166, in from_urdf_file
    links = URDF_utils.get_urdf_parameters(urdf_file, base_elements=base_elements, last_link_vector=last_link_vector, base_element_type=base_element_type)

  File "C:\Users\alexa\Anaconda3\lib\site-packages\ikpy\URDF_utils.py", line 127, in get_urdf_parameters
    tree = ET.parse(urdf_file)

  File "C:\Users\alexa\Anaconda3\lib\xml\etree\ElementTree.py", line 1197, in parse
    tree.parse(source, parser)

  File "C:\Users\alexa\Anaconda3\lib\xml\etree\ElementTree.py", line 598, in parse
    self._root = parser._parse_whole(source)

  File "<string>", line unknown
ParseError: not well-formed (invalid token): line 45, column 93

**Thanks a lot for your help**

Phylliade, 2020-01-24:
Hello @Alexa875 , 

Could you provide the URDF file you're using?

Alexa875, 2020-01-24:
Hello,
The poppy ergo .urdf

Phylliade, 2020-01-25:
This is note the URDF file actually :)

You may find the right file below: https://raw.githubusercontent.com/Phylliade/ikpy/master/resources/poppy_ergo.URDF (just copy paste the content of this file)

closed 2020-01-25 by Phylliade

Alexa875, 2020-01-25:
Work perfectly

Cela a très bien fonctionné avec le fichier en ressources. 
Désolé je n'avais pas saisi qu'il fallait s'appuyer sur ce dossier.
Merci pour votre aide précieuse et votre travail livré à la communauté. 
Cordialement