Phylliade/ikpy #76: Transform matrix, whom to whom?

leemengwei, 2020-03-11:
Hi, thanks for your work. Great one, I like it. I'm doing some deeplearning research, thus I'm not quite familiar with robotics. Sorry for my innocent,
So here's my problem about what is 'real_frame' in code, does that mean to transform from end-effector to old one? (I think it is, but not sure)
Take two frame for example, world_frame and new_frame,  when joint is given:
`<joint                 
  name="joint1"        
  type="revolute">     
  <origin              
    xyz="0 -0.43 0"    
    rpy="1.5708 0 0" />        <-------rotation around axis X?    counter-clock wise? I guess?
  <parent              
    link="base_link" />
  <child               
    link="link1" />    
  <axis                
    xyz="0 0 1" />     
  <limit               
    lower="0"          
    upper="0"          
    effort="0"         
    velocity="0" />    
</joint>               
`
then returned two frames are:
[[[ 1.    0.    0.    0.  ]
  [ 0.    1.    0.    0.  ]
  [ 0.    0.    1.    0.  ]
  [ 0.    0.    0.    1.  ]]

 [[ 1.    0.    0.    0.  ]
  [ 0.   -0.   -1.   -0.43]
  [ 0.    1.   -0.    0.  ]
  [ 0.    0.    0.    1.  ]]

Phylliade, 2020-03-16:
Hello @leemengwei,

could you paste the code that procuded these outputs?

leemengwei, 2020-03-17:
Ops the codes are gone, I've modified them, yet my uncertain remains the same. The first matrix I gave out
[[[ 1. 0. 0. 0. ]
[ 0. 1. 0. 0. ]
[ 0. 0. 1. 0. ]
[ 0. 0. 0. 1. ]]
is something like base frame, world frame or say, I guess.
The second matrix
[[ 1. 0. 0. 0. ]
[ 0. -0. -1. -0.43]
[ 0. 1. -0. 0. ]
[ 0. 0. 0. 1. ]]
is frame I got after _forward kinematic_ , something I got by:
`new_frame = np.array(my_chain.forward_kinematics(six_axis)`

what does the second matrix mean? Is't transformation from base to new, or from new to base? (I think is the new to base, am I right?)

Phylliade, 2020-03-22:
This matrix is a position encoded as homogeneous coordinates;

```
[[ 1. 0. 0. 0. ]
[ 0. -0. -1. -0.43]
[ 0. 1. -0. 0. ]
[ 0. 0. 0. 1. ]]
```

With this, you can see that your position is `matrix[3, :3]`, so here the position of your end-effector is `[0, 0.43, 0]`

closed 2020-03-22 by Phylliade