Phylliade/ikpy #77: Exception "RuntimeError: generator raised StopIteration" with Python3.7

cjlux, 2020-04-11:
Python 3.7 has change the behavior of generators : when they reach the end now they emit a RuntimeError exception... 
In the file URDF_utils.py there are two occurences of next(...) that can raise RuntimeError exception, lines 91 and 103.

Line 103 "chain = list(next(it) for it in itertools.cycle(iters))" can be replaced by:
```
chain = []
    for it in itertools.cycle(iters):
        try:
            item = next(it)
        except:
            break
        chain.append(item)
```

Phylliade, 2020-04-25:
Good catch @cjlux !

Actually this function seems deprecated and should be removed (seems it's not used  anywhere).

In which context did you use it?

cjlux, 2020-04-25:
We (Yoan Mollard, Thibault Desprez and myself) are presently updating the _raspoppyfication_ process (installing Poppy configuration on a RPi4 under raspbian buster) :  the IKpy module is used at start-up by the robot "Ergo Junior"  and I found also this bug when using IKpy for the robot "Poppy Torso" to make the robot catch objects recognized by tits camera.

cjlux, 2020-04-25:
sorry closing was  accidental...

Phylliade, 2020-04-25:
In order to use URDF, the function `get_urdf_parameters` should be used instead.

Phylliade, 2020-04-25:
On the other hand, if you explicitly need it, we can move it to a "deprecated" module (and fix the original issue), instead of just removing it.

cjlux, 2020-05-07:
After investigation I found this : the function **get_chain_from_joints** defined in the file _URDF_utils.py_ is called at line 25 of the file _ik.py_ from the module _pypot.creature_ : 
`25    chain_elements = get_chain_from_joints(poppy.urdf_file, [m.name for m in motors])`
and the liste `chain_elements` contains for example : 
`['base', 'abs_z', 'spine', 'bust_y', 'bust_motors', 'bust_x', 'chest', 'l_shoulder_y', 'l_shoulder', 'l_shoulder_x', 'l_shoulder_motor', 'l_arm_z', 'l_upper_arm', 'l_elbow_y']`

Some lines later in the file _ik.py_  the list `chain_elements` is given as argument to the call :
```
30         chain = cls.from_urdf_file(poppy.urdf_file,
31                                    base_elements=chain_elements,
32                                    last_link_vector=tip,
33                                    active_links_mask=activ)`
```
and the function `from_urdf_file` (in the file _chain.py_ from the module IKpy) calls itself the function **get_urdf_parameters** with among others the argument ` base_elements=chain_elements`...

So if you plan to deprecate the function **get_chain_from_joints** the question is : how to generate the list `chain_elements` ? 
Or a related question is : what kind of information in details is required for the argument `base_elements` of the **get_urdf_parameters** function ?

Phylliade, 2020-05-24:
Actually this function seems useful in the way it's used, since it permits to get a list of links+joints from a list of joints. 

And this use-case can be appear in the pypot creatures context: Since IKp's URDF parser requires a list of links+joints, and pypot works with a list of joints, this function makes the link between the two

commit 8fe8cb04 (2020-05-27): Fix #77
PR #88 (2020-05-27): Close `get_chain_from_joints` broken with Python 3.7
Fix #77
closed 2020-05-27 by Phylliade