Phylliade/ikpy #85: Value Error inverse_kinematics

HaffmannGG, 2020-05-13:
Following the getting started i got an issue with Setting the position of your chain (aka Inverse Kinematics, aka IK). 

File "C:\Users\Jonas\untitled1.py", line 46, in <module>
    frame_target[:3,:3] = target_position

ValueError: cannot copy sequence with size 4 to array axis with dimension 3

After changing the line to "frame_target = target_position" the error is gone and the presented results appear in the console. Am i just a noob or is this broken?

Phylliade, 2020-05-13:
Hello @HaffmannGG, can you provide the full code snippet that appears?

(i.e. what is `target_position`?)

HaffmannGG, 2020-05-13:
`from ikpy.chain import Chain
from ikpy.link import OriginLink, URDFLink

my_chain = Chain.from_urdf_file("C:\\*\\*\\*\\ikpy-master\\ikpy-master\\resources\\poppy_ergo.URDF")

my_chain.inverse_kinematics([[1, 0, 0, 2],
                    [0, 1, 0, 2],
                    [0, 0, 1, 2],
                    [0, 0, 0, 1]])`

I use Anaconda and Spyder as my env.

runfile('C:/*/*/untitled1.py', wdir='C:/*/*')
Traceback (most recent call last):

  File "C:\*\*\untitled1.py", line 16, in <module>
    [0, 0, 0, 1]])

  File "C:\Users\Jonas\anaconda3\lib\site-packages\ikpy\chain.py", line 138, in inverse_kinematics
    frame_target[:3, 3] = target_position

ValueError: cannot copy sequence with size 4 to array axis with dimension 3

Phylliade, 2020-05-13:
It seems your `target_position` is not a three dimensional vector, as it should be

HaffmannGG, 2020-05-13:
The inserted matrix is 4 by 4 and if i call "inverse_kinematics" it takes the only argument as target_position.  Am i missing something?

HaffmannGG, 2020-05-13:
My bad. I did follow the Getting Started from your wiki page (https://github.com/Phylliade/ikpy/wiki). In these steps you call the inverse kinematics function with a 4x4 matrix. I checked your Quickstart file and in this case you use a 1x3 vector. Everythin is working fine.

closed 2020-05-13 by HaffmannGG

alxy, 2020-06-14:
I have a actually the same same issue and a script at hand which passes a 4x4 matrix to the inverse_kinematics function as described here: https://github.com/Phylliade/ikpy/wiki#setting-the-position-of-your-chain-aka-inverse-kinematics-aka-ik

Has this been possible in a former version of the library and now it only accepts a three valued vector? In this case, you should update the wiki ;) @Phylliade 
If I see that correctly the orientation is now handled via the second argument of that function, right?

HaffmannGG, 2020-06-14:
@alxy you are right. This link is outdated.

Phylliade, 2020-06-20:
Hello @HaffmannGG @alxy, thanks for pointing this out, it's now fixed!