Phylliade/ikpy #91: BVH parser

gemlongman, 2020-07-13:
Is there a simple way to create a chain with BVH motion files？In fact it will be perfect to add a BVH parser.

Phylliade, 2020-07-14:
Hello @gemlongman,

I'm not familiar with BVH motion capture files: do you know how exactly they encode their data? (And would you have an example?)

gemlongman, 2020-07-15:
Here is a BVH file in the zip which could be found more in https://sites.google.com/a/cgspeed.com/cgspeed/motion-capture/daz-friendly-release, they can be imported by Motion Builder to see motion. Meanwhile, there is some BVH python parser such as https://pypi.org/project/bvh/, but it will take lots of work to adapt.
[Running  Jump To Standing Idle.zip](https://github.com/Phylliade/ikpy/files/4922618/Running.Jump.To.Standing.Idle.zip)

Phylliade, 2020-08-02:
After some investigations I'm joining you in the fact that's a super good idea!

The BVH file's HIERARCHY section is very close to the definition of a URDF, so this shouldn't be so difficult to rewrite a parser, even from scratch.
I've found a good spec here: https://research.cs.wisc.edu/graphics/Courses/cs-838-1999/Jeff/BVH.html

Regarding the features, am I right to suppose we will need actually two features?

* Creating a chain from a BVH HIERARCHY section
* Importing a forward kinematics trajectory from a BVH MOTION section

gemlongman, 2020-08-04:
yeah~you are right

andrewnc, 2024-06-22:
I'm thinking about picking this up, mostly to work with BVH in Mujoco - but would gladly provide code here too. Was any more progress made outside of this issue?

andrewnc, 2024-06-24:
`import xml.etree.ElementTree as ET
import xml.dom.minidom
import numpy as np

class BVHtoURDF:
    def __init__(self, bvh_file):
        self.bvh_file = bvh_file
        self.hierarchy = {}
        self.urdf_tree = None
        self.joint_counter = 0
        self.lowest_point = np.array([0, 0, 0])
        self.root_offset = np.array([0, 0, 0])

    def parse_bvh(self):
        with open(self.bvh_file, 'r') as f:
            content = f.read()
        
        hierarchy_section = content.split('MOTION:')[0]
        self.parse_hierarchy(hierarchy_section)

    def parse_hierarchy(self, hierarchy_section):
        lines = hierarchy_section.strip().split('\n')
        stack = []
        current_joint = None

        for line in lines:
            parts = line.strip().split()
            if not parts:
                continue

            if parts[0] == 'ROOT' or parts[0] == 'JOINT':
                joint_name = parts[1]
                self.joint_counter += 1
                current_joint = {
                    'name': joint_name,
                    'id': self.joint_counter,
                    'offset': None,
                    'channels': [],
                    'children': []
                }
                if stack:
                    stack[-1]['children'].append(current_joint)
                stack.append(current_joint)

            elif parts[0] == 'OFFSET':
                current_joint['offset'] = [float(x) for x in parts[1:4]]

            elif parts[0] == 'CHANNELS':
                current_joint['channels'] = parts[2:]

            elif parts[0] == 'End':
                self.joint_counter += 1
                end_site = {
                    'name': f"{current_joint['name']}_end",
                    'id': self.joint_counter,
                    'offset': None,
                    'channels': [],
                    'children': []
                }
                current_joint['children'].append(end_site)
                stack.append(end_site)

            elif parts[0] == '}':
                if stack:
                    completed_joint = stack.pop()
                    if not stack:
                        self.hierarchy = completed_joint

        self.calculate_lowest_point(self.hierarchy, np.array([0, 0, 0]))
        self.root_offset = np.array([0, -self.lowest_point[1], 0])
[... 96 lines cut ...]
        izz = (1/2) * mass * radius**2
        return ixx, iyy, izz

    @staticmethod
    def rotate_vector(vector, angle):
        rotation_matrix = np.array([
            [1, 0, 0],
            [0, np.cos(angle), -np.sin(angle)],
            [0, np.sin(angle), np.cos(angle)]
        ])
        return np.dot(rotation_matrix, vector)

    def save_urdf(self, output_file):
        xml_str = ET.tostring(self.urdf_tree, encoding='utf-8')
        dom = xml.dom.minidom.parseString(xml_str)
        pretty_xml_str = dom.toprettyxml(indent="  ")
        
        with open(output_file, 'w') as f:
            f.write(pretty_xml_str)

# Usage
parser = BVHtoURDF('input.bvh')
parser.parse_bvh()
parser.create_urdf(rotation_angle=np.pi/2)  # Rotate 90 degrees around X-axis
parser.save_urdf('output.urdf')`

Phylliade, 2026-10-26:
Shipped in 4.2.0: `Chain.from_bvh_file(file, base_elements=["Hips", "LeftHand"], scale=0.01)` builds a chain from any path of a BVH skeleton (the joints between two names are filled in), and `ikpy.bvh.BVH.get_bvh_motion` returns the MOTION section as joint vectors of that chain, ready for `forward_kinematics`. Each channel becomes one link, so the rotation order of every joint is reproduced exactly; position channels are handled on any joint (Mixamo style exports included). The result was checked against pybvh's forward kinematics on CMU, Mixamo, mocapbank, MakeHuman and SnapMoGen captures, including the file posted in this thread. See the "Motion capture (BVH) Support" section of the README.

closed 2026-10-26 by Phylliade