Phylliade/ikpy #96: When will the Prismatic Joint added?

AhmetMericOzcan, 2020-11-01:
I really like to use a prismatic joint solution in my project. Could you please tell me a date for primsatic joint support? Or could you please tell me how it can be done?

Phylliade, 2020-11-08:
Hello @AhmetMericOzcan,

I have a prototype but lacks a meaningful URDF to make a real-world try;

Can you provide a URDF with a prismatic joint?

AhmetMericOzcan, 2020-11-09:
Hello in this link there is a prismatic joint prototype with visualization. Hope this helps. If this doesnt help please contact me again. I might look for futher solutions.

https://mymodelrobot.appspot.com/5629499534213120

AhmetMericOzcan, 2020-11-09:
<?xml version="1.0" ?>
<robot name="urdf_robot">
  <link name="baseLink">
    <inertial>
      <origin rpy="0 0 0" xyz="0 0 0"/>
       <mass value="0.0"/>
       <inertia ixx="0" ixy="0" ixz="0" iyy="0" iyz="0" izz="0"/>
    </inertial>
    <visual>
      <origin rpy="0 0 0" xyz="0 0 0"/>
      <geometry>
      <box size="0.1 .1 .1"/>
      </geometry>
    </visual>
    <collision>
      <origin rpy="0 0 0" xyz="0 0 0"/>
      <geometry>
	 <box size="0.1 .1 .1"/>
      </geometry>
    </collision>
  </link>
   <link name="childA">
    <inertial>
      <origin rpy="0 0 0" xyz="0 1 -1"/>
      <mass value="10.0"/>
      <inertia ixx="1" ixy="0" ixz="0" iyy="1" iyz="0" izz="1"/>
    </inertial>
    <visual>
      <origin rpy="0 0 0" xyz="0 1 -1"/>
      <geometry>
        <box size="0.1 .1 .1"/>
      </geometry>
    </visual>
    <collision>
      <origin rpy="0 0 0" xyz="0 1 -1"/>
      <geometry>
        <box size="0.1 .1 .1"/>
      </geometry>
    </collision>
  </link>
   <joint name="joint_baseLink_childA" type="prismatic">
    <parent link="baseLink"/>
    <child link="childA"/>
    <origin xyz="0 0 1.0"/>
	<axis xyz="0 0 1"/>
	<limit lower="-0.1" upper="0.2" />
  </joint>
 
</robot>

I found a simple prismatic joint model. Hope this helps. You can visualize it on the web page below:

https://mymodelrobot.appspot.com/5629499534213120

oxbown, 2020-11-17:
Really interested as well. Is there any update on Prismatic joints?

Klausstaler, 2021-02-08:
@oxbown @AhmetMericOzcan I had this issue myself and solved it for the URDFLink and the URDF to joints parser. If you want to take a look or try it out, checkout my fork https://github.com/Klausstaler/ikpy

alters-mit, 2021-03-19:
@Klausstaler I looked over your code but I'm not sure how to implement a prismatic joint. Could you provide a little example code of how to set up the `Link` constructor parameters? I see a `is_revolute` parameter but I don't understand how to define the prismatic limits.

Klausstaler, 2021-03-19:
@alters-mit The prismatic limits are defined in the same way you'd define limits for revolute joints. Instead of radians, they are now in meters though. As in the example below, you include a `<limit>` tag together with the lower and upper bound for each joint you have to define limits for.  The exact parsing can be found [here](https://github.com/Klausstaler/ikpy/blob/master/src/ikpy/urdf/URDF.py#L224). 
```
<?xml version="1.0"?>
<robot name="robot">
    <link name="base_link">
    </link>
    <joint name="base_link_base_robot_joint" type="fixed">
        <parent link="base_link"/>
        <child link="base_robot"/>
        <origin xyz="0.0 0.0 0.0" rpy="0 0 0"/>
    </joint>
    <link name="base_robot">
    </link>
    <link name="slider">
    </link>
    <joint name="base_robot_slider_joint" type="fixed">
        <parent link="base_robot"/>
        <child link="slider"/>
        <origin xyz="0 0 0" rpy="0 -1.5707953 0"/>
    </joint>
    <joint name="linear_actuator" type="prismatic">
        <parent link="slider"/>
        <child link="slider_arm"/>
        <axis xyz="-1 -2.368514e-06 4.1005696e-06"/>
        <origin xyz="0 0 0.02" rpy="-0.5237906 1.5707916 1.0470034"/>
        <limit effort="180" lower="0" upper="0.4" velocity="10"/>
    </joint>
    <link name="slider_arm">
        <visual>
            <origin xyz="0.032 0.023 0.07" rpy="1.5707963 0 0"/>
            <geometry>
                <cylinder radius="0.036" length="0.046"/>
            </geometry>
        </visual>
        <collision>
            <origin xyz="0.032 0.023 0.07" rpy="1.5707963 0 0"/>
            <geometry>
                <cylinder radius="0.036" length="0.046"/>
            </geometry>
        </collision>
    </link>
</robot>
```

alters-mit, 2021-03-19:
Thanks. Is there a way to instantiate a prismatic joint from a class constructor? For my particular project, I'm defining a list of `Link` objects in a chain but I'm not actually reading from a urdf file.

alters-mit, 2021-03-23:
@Klausstaler I forked off of your repo but found problems with applying the axis of a prismatic joint as well as distinguishing prismatic joints from fixed joints. This seems to work much better: https://github.com/alters-mit/ikpy/blob/53e6069e800927807befc6a10a411de304b4b634/src/ikpy/link.py#L179-L183

closed 2021-07-10 by Phylliade

Phylliade, 2021-07-10:
Hello @AhmetMericOzcan, support for prismatic joints has landed on master, with the associated release `v3.2`. 

Enjoy!
And tell me if this answer your need!

PR #109 (2021-07-10): Add support for prismatic joint
Add full support for prismatic joints:
* type="prismatic" in URDF
* forward kinematics
* bounds
* prismatic joints axes are plotted in dotted lines, where revolute joints are plotted in solid lines

Fixes #96 #104 

Ex on URDF provided by https://github.com/AhmetMericOzcan. The prismatic joint is the dotted one:

[image]

When sliding the prismatic joint, we see it moving:
[image]

# Additional features

* Support for arbitrary joint types in the URDF
* Fixed a bug in URDF parsing where `fixed` joints could be considered as revolute
* Support for customizable op [... cut]