Phylliade/ikpy #101: Tutorial Error

MADONOKOUKI, 2020-12-14:
Hi, my name is koki.
I got an error like that. I use anaconda3-5.3.1 and Python 3.5.6.
I want to know why was it error happened?
`>>> import ikpy
>>> my_chain = ikpy.chain.Chain.from_urdf_file("model.urdf")
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
AttributeError: module 'ikpy' has no attribute 'chain'`

Benoit-LdL, 2020-12-14:
Same issue here, I can't seem to get any submodule of ikpy to work. I'm running version 3.0.1 with python 2.7.12...
Tried some of the test scripts without any luck
`
from ikpy.link import OriginLink, URDFLink
--> ImportError: No module named link

from ikpy.chain import Chain
--> ImportError: No module named chain

import ikpy.utils.plot as plot_utils
--> ImportError: No module named utils.plot
`

Thanks in advance for your help!
Kind regards,
Benoît

Phylliade, 2021-01-17:
Hello @MADONOKOUKI @Benoit-LdL,

How are you installing IKPy?
(On my side, i'm not able to reproduce this issue)

Besides, @MADONOKOUKI, what version of IKPy are you using?

KyleORourke2, 2021-01-28:
I'm having the same issue. I'm using the latest version of ikpy v3.1 with Python 3.8.3. I've tried both 'pip install ikpy' and from source with the same error: 

> module 'ikpy' has no attribute 'chain'

Zju-George, 2021-02-03:
@Phylliade @Benoit-LdL @MADONOKOUKI @KyleORourke2 Check out this: https://stackoverflow.com/questions/8899198/module-has-no-attribute
In pyik the submodule is not directory but pyfile so it is not imported automatically.
A quick solution, add `import` in __init__.py like this
[image]
Then it will work.
[image]

Piit3r, 2021-02-16:
Same here in local.
I also tested on google colab :
I imported de Quickstart file. Just run `pip install ikpy` on top and execute

closed 2021-04-26 by stale[bot]

Phylliade, 2021-07-07:
Hello @MADONOKOUKI, this has been fixed in master, you call pull the latest ipynb file and this will work.
No need to update IKPy.

commit cd6775ac (2021-07-07): Fix tutorial - fixes #101