Phylliade/ikpy #103: Option to output as Quaternions/Euler angles/Vectors

Toxic-Cookie, 2021-01-18:
The current output is homogeneous coordinates. For generalized use of this software, it would make sense to have an option for the outputs to be in the context of 3D engines like Blender and Unity.

Phylliade, 2021-07-08:
Hello @Toxic-Cookie ,

This is indeed a good feature, I'm pinning it, and will see when this could be done.

In the meantime, you can use the Scipy primitives, as defined here:
https://stackoverflow.com/questions/53033620/how-to-convert-euler-angles-to-quaternions-and-get-the-same-euler-angles-back-fr

(Second answer)

Toxic-Cookie, 2021-07-08:
Took a little while but I'm glad you're looking into it!

Phylliade, 2021-07-08:
Are the Scipy primitives to convert to quaternions ok for you?

Toxic-Cookie, 2021-07-14:
> Are the Scipy primitives to convert to quaternions ok for you?

Sorry for the late reply! Github didn't bother notifying me. Yeah quaternions would be fantastic!

Phylliade, 2026-10-26:
Shipped in 4.2.0: `ikpy.utils.geometry` now has `rotation_matrix_to_quaternion` / `quaternion_to_rotation_matrix` (scalar-last `(x, y, z, w)` by default as in ROS and Unity, `scalar_first=True` for Blender's `(w, x, y, z)`), `rotation_matrix_to_rpy` (inverse of `rpy_matrix`, URDF convention) and `rotation_matrix_to_euler` / `euler_to_rotation_matrix` for any axis sequence. They take the 4x4 output of `forward_kinematics` directly, and the matrix they return can be passed as `target_orientation` with `orientation_mode="all"`. See the "Orientation conversions" section of the README.

closed 2026-10-26 by Phylliade