Phylliade/ikpy #106: Create Chain method won't make chain from URDF file

jurka14, 2021-03-04:
Hi @Phylliade , I have been having some problems while trying to use the  .from_urdf_file() method with my URDF file. When using the tutorial file poppy_ergo.URDF, it works fine, but with my file it does not actually create a chain, but a simple base_link fill in.

Here is the python code I'm using:
`import ikpy
import ikpy.chain
import numpy as np

my_chain = ikpy.chain.Chain.from_urdf_file("C:\\Users\\jurka\\Desktop\\Rameno_sestava.SLDASM.URDF")

target_position = [ 0.1, -0.2, 0.1]

print("The angles of each joints are : ", my_chain.inverse_kinematics(target_position))`

And here is the output:
`The angles of each joints are :  [0.]`

Also here is my URDF file:

[Rameno_sestava.SLDASM.zip](https://github.com/Phylliade/ikpy/files/6076311/Rameno_sestava.SLDASM.zip)

Phylliade, 2021-07-08:
Hello @jurka14,

This is very strange, since this URDF is tested frequently and works.

Can you compare your code to this one:

https://www.github.com/Phylliade/ikpy/tree/master/tests%2Ftest_poppy_robots.py
?

Phylliade, 2021-07-10:
Hello @jurka14,

The error is coming from the fact that you need to specify your root element of your robot using the `base_elements` argument

If you do: 

```
chain1 = chain.Chain.from_urdf_file(
    "Rameno_sestava.SLDASM.urdf", base_elements=["base_Link"])
print(chain1)
# The resulting chain1 is:
# Kinematic chain name=chain links=['Base link', 'Rotating_base_Joint', 'First_arm_Joint', 'Second_arm_Joint', 'Barrel_part_Joint', #'End_point_Joint'] active_links=[ True  True  True  True  True  True]
```

This works!

In parallel, IKPy was silently failing when you provided a wrong `base_elements` (or were using the default one when it didn't fit, as in your case). This has been fixed in `v3.2` and IKPy will raise an explicit error

closed 2021-07-10 by Phylliade