Phylliade/ikpy #108: Changing the optimization algorithm

ger01d, 2021-03-13:
Is it possible to change the optimization algorithm with a function call, or is it only possible to statically change line 134 in _inverse_kinematics.py_?
With _L-BFGS-B_ and joint boundaries I noticed a jumpy behavior in the end effector positions of the kinematic model, so I changed it to _SLSQP_.

I also think that my model seems to be correctly defined:
[](https://github.com/ger01d/kinematic-model-opencat)

Phylliade, 2021-07-08:
Hello @ger01d,

Did changing the optimization method to  SLSQP solve your issue?

In the meantime, I will add a method to select the optimization algorithm in the constructor

ger01d, 2021-07-08:
Hi @Phylliade,

yes, changing to SLSQP solved the problem. Thanks for bringing this feature into the code.

closed 2021-07-10 by Phylliade

Phylliade, 2021-07-10:
Hello @ger01d, 
Support for customizable optimization algorithms has landed on master and on `v3.2`. Just use the `optimization_method` of the `Chain.inverse_kinematics`.

MarcDcls, 2022-04-25:
Hi @Phylliade, 

It seems that the argument `optimization_method` is never used in `inverse_kinematic_optimization` , can we really change the optimization algorithm through this argument ?

Phylliade, 2022-04-26:
Hello @MarcDcls , yep you're right, this parameter is not used anymore

Phylliade, 2022-04-26:
Do you need access to the old optimization method, based on L-BFGS-B?

In https://github.com/Phylliade/ikpy/issues/123, access to L-BFGS-B is required because the new optimization method it breaks performances on some cases (to be dug)

MarcDcls, 2022-04-29:
Hi @Phylliade, 

I wanted to test several optimization methods to find the one that fit the best my problem. I've managed to do it by myself so I don't have issue anymore. Thank for your help

PR #109 (2021-07-10): Add support for prismatic joint
Add full support for prismatic joints:
* type="prismatic" in URDF
* forward kinematics
* bounds
* prismatic joints axes are plotted in dotted lines, where revolute joints are plotted in solid lines

Fixes #96 #104 

Ex on URDF provided by https://github.com/AhmetMericOzcan. The prismatic joint is the dotted one:

[image]

When sliding the prismatic joint, we see it moving:
[image]

# Additional features

* Support for arbitrary joint types in the URDF
* Fixed a bug in URDF parsing where `fixed` joints could be considered as revolute
* Support for customizable op [... cut]