Phylliade/ikpy #116: Need help to create a chain from global placements

FlachyJoe, 2022-01-11:
Hi,
I'm trying to use ikpy with FreeCAD and can't find how to create a chain from object placements.

Placements are global coordinates and decomposable in Base (translation from origin) and Rotation (quaternion).
I think I get a link origin_translation by subtraction of its Base by the last link Base but can't figure how to get the origin_rotation.

Thanks !

Phylliade, 2022-01-13:
Hello,

Do you know the value of your `origin_orientation`?

FlachyJoe, 2022-01-13:
Basically I only know objects placement in global coordinates. I understood a link origin_rotation is defined in local coordinates so no: I don't know each link origin_rotation.
An other question torments me: is axis defined in local coordinates too ?

Phylliade, 2022-01-17:
Ok I get it.

In this case you don't care about origin_translation or origin_rotation.
These are positional attributes of your link at the origin of the robot (e.g. the table)

In your case, you want to position the other links of the robot and that's easy:

1. Create a robot using a chain of links, indépendant of the position
2. Apply inverse kinematics on the robot. In order to do so, you effectively need to transform your Pos+Yaw/Pitch/Roll coordinates into a transformation matrix. This can be done easily using the following function: https://github.com/Phylliade/ikpy/blob/master/src/ikpy/utils/geometry.py#L56

FlachyJoe, 2022-01-17:
1. But I can't create a chain of links with all null origins and rotations, it has no sense ! As I only know the objects global placement and relative rotation axis I have to compute the links data from them.
2. I already use inverse kinematics to set the robot in this position but it only works from an aligned state. If I have a robot in the state shown I don't know how to put it back (think I can't save the chain data).

FlachyJoe, 2022-01-18:
>  without any position or orientation
I'm not agree as you use _ikpy/resources/poppy_ergo.URDF_ in which we find such data:
`<link
    name="base_link">
    <inertial>
      <origin
        xyz="-0.00619082439227047 1.87708924758407E-10 0.0111397401808713"
        rpy="0 0 0" />`
parsed by https://github.com/Phylliade/ikpy/blob/3966be998ce0ece1d1068dd11b5f2c42b172cf30/src/ikpy/urdf/URDF.py#L241
so origin_translation and origin_orientation **are** initialized.

I still argue that building a string without these values makes no sense. Firstly because joint angle obviously depends on link length.

Phylliade, 2022-01-19:
Yep you're right, that's just a misanderstanding.

To recap: 

* You have on one hand your **chains's geometry**, where each link has a defined geometry, such as `<link name="base_link"><origin xyz="-0.00619082439227047 1.87708924758407E-10 0.0111397401808713" rpy="0 0 0" />` That's all you need to create a chain. It is something you bring by yourself when using IKPy
* You move your chain using **actuators positions** (e.g. angles), using IKPy forward and inverse kinematics functions
* And get **final positions**

From what i get of your issue is that you want to infer chain geometry from **final positions**. 
The issue is that it's only one particular final position, and you won't be able to infer **rotation axes** from the final positions (you can infer positions and orientations, but not the rotation axes).

Unfortunately IKPy won't help you here. 
The simplest solution by far is to ask FreeCAD to export your chain's geometry, for example to export a URDF file, as in https://answers.ros.org/question/355109/create-a-urdfxacro-file-from-model-of-freecad/

With this URDF file, you can use IKPy on it, as described in https://github.com/Phylliade/ikpy/blob/master/tutorials/Quickstart.ipynb

FlachyJoe, 2022-01-19:
Solved.
FreeCAD has placement calculus implemented and I found how to get origin relative to another.
```
[…]
	URDFLink(
		"Arm3",
		(obj('Arm2').Placement.inverse()*obj('Arm3').Placement).Base, # origin translation
		rpy((obj('Arm2').Placement.inverse()*obj('Arm3').Placement).Rotation), # origin rotation
		[0, 1, 0], # rotation axis
		None,  # translation axis
		(None, None),  # bounds
	)
[…]
```

closed 2022-01-19 by FlachyJoe

Phylliade, 2022-01-19:
Great!