Phylliade/ikpy #122: Not working with full referential (orientation mode "all") with unity orientation

Miker2808, 2022-03-15:
Tried quaternions and eulers, in any possible combination, it always works on one axis and died completely in another when the orientation comes from unity, (both quaternions and eulers) if you have a suggestion or would like to make it a possibility to insert quaternions instead of rotation matrixes it would be great, so far it seems the IK works only for position and not orientation.
I attempted rotation translation using scipy but with no luck.

Phylliade, 2022-03-17:
Hello @Miker2808, can you copy paste the code you're using?

Miker2808, 2022-03-21:
Sorry if my issue was a little aggressive, I really lost my mind over this orientation. unity conflicting with my URDF made it really difficult with shifting orientation. 
I'm including the full script, it communicates using ROS2 (ROS2 coms from unity straight away as Vector3 or Quaternion) depending on the script, I can include a quaternion script as well. 
my Unity is Y axis is up, while the URDF is Z axis is up.

Sidenode: the wiki with the "full orientation" has a "sample" which says "Look at the sample below" and right under it there is nothing. 

```

import numpy as np

from ikpy.chain import Chain
from ikpy.link import OriginLink, URDFLink
from ikpy.inverse_kinematics import inverse_kinematic_optimization
import ikpy.utils.plot as plot_utils
import matplotlib.pyplot as plt
from urllib3 import Retry

import rclpy
from rclpy.node import Node
from sensor_msgs.msg import JointState
from geometry_msgs.msg import Vector3

from scipy.spatial.transform import Rotation

hdt_chain = Chain(name="Hdt_Adroit_Chain", active_links_mask=[1,0,1,0,0,1,0,0,1,0,1,1,0], links=[
    #URDFLink(
    #    name="integration_kit_joint",
    #    origin_translation=[0.0, 0.0, 0.0],
    #    origin_orientation=[0.0, 0.0 ,0.0],
    #    joint_type="fixed"
    #),
    URDFLink(
        name="joint1",
        origin_translation=[-0.1, 0.0, 0.25],
        origin_orientation=[0.0, 0.0 ,0.0],
        rotation=[0,0,1],
        bounds=[-2.7,2.7],
        joint_type="revolute"
    ),
    URDFLink(
        name="joint1f",
        origin_translation=[0.0, 0.0, 0.09217],
        origin_orientation=[0.0, 0.0 ,0.0],
        joint_type="fixed"
    ),
    URDFLink(
        name="joint2",
        origin_translation=[0.0, 0.0, 0.07],
        origin_orientation=[0.0, 0.0 ,0.0],
        rotation=[1,0,0],
        bounds=[-2.0,2.0],
        joint_type="revolute"
    ),
    URDFLink(
        name="joint2f",
        origin_translation=[0.0, 0.0, 0.077],
        origin_orientation=[0.0, 0.0 ,0.0],
        joint_type="fixed"
    ),
    URDFLink(
        name="joint3f",
        origin_translation=[0.0, 0.0, 0.2530],
        origin_orientation=[0.0, 0.0 ,0.0],
        joint_type="fixed"
    ),
    URDFLink(
        name="joint4",
        origin_translation=[0.0, 0.0, 0.07],
        origin_orientation=[0.0, 0.0 ,0.0],
        rotation=[1,0,0],
[... 165 lines cut ...]
        self.orientation_y_msg = msg
    
    def orientation_z_callback(self,msg):
        self.orientation_z_msg = msg
        

            

joints_error_list = [0.0, 0.00, 0.0, -0.30, 0.0, 0.0, 0.00, 0.0, 0.0, 0.00, 0.0, -0.35, 0.00, 0.0]
count_joint = [0,1,0,1,0,0,1,0,0,1,0,1,1,0]
def main(args=None):

    
    rclpy.init(args=args)

    joint_publisher = AdroitIKPublisher(target_velocity=1.0)

    rclpy.spin(joint_publisher)

    joint_publisher.destroy_node()
    rclpy.shutdown()

if __name__ == "__main__":
    main()
```

Phylliade, 2022-04-05:
Here is a working example of the full orientation mode:

```
def test_orientation_full_frame(baxter_left_arm):
    # Found by exploring the reachable values
    target_position = [0.1, 0.5, -0.1]
    target_orientation = [[0,0,1], [1,0,0], [0,1,0]]

    # Begin to place the arm in the right position
    ik = baxter_left_arm.inverse_kinematics(target_position)
    ik = baxter_left_arm.inverse_kinematics(target_position, target_orientation, initial_position=ik, orientation_mode='all')

    position = baxter_left_arm.forward_kinematics(ik)[:3, 3]
    orientation = baxter_left_arm.forward_kinematics(ik)[:3, :3]

    # Check
    np.testing.assert_almost_equal(orientation, target_orientation, decimal=5)
    np.testing.assert_almost_equal(position, target_position, decimal=5)
```

Phylliade, 2022-04-05:
Taken from here: https://github.com/Phylliade/ikpy/blob/master/tests/test_orientation.py#L26

closed 2022-07-30 by stale[bot]