Phylliade/ikpy #124: Trouble creating chain with URDFlink

nickredsox, 2022-04-03:
Hello,

I am having a problem trying to create a chain from a URDF link. Right now I am trying to model a 4 dof robot similar to Trossen PincherX (https://www.trossenrobotics.com/pincherx-100-robot-arm.aspx) where I have a base that rotates about z axis and three other joints that rotate about another axis.

I have been trying to create with URDFlink using the example but I think I must be missing something (Right now I am only using 3 links to troubleshoot ) but I can only get a rotation about the y axis and not the z. For the inverse kinematics if I set the position to be (5,0,5) (x,y,z) the robot can reach that point and gives me joint angles. However, if I set the position to (0,5,5), the robot just returns joint angles of zeros. If I manually set the join position  with 

`my_chain.forward_kinematics([0, 1.5708, 1.5708, 0])`

 I do get a transformation matrix that shows I am at (0,5,5)

[[ 1.34924357e-11 -1.00000000e+00 -3.67320510e-06 -1.83660255e-05]
 [-3.67320510e-06 -3.67320510e-06  1.00000000e+00  5.00000000e+00]
 [-1.00000000e+00  0.00000000e+00 -3.67320510e-06  4.99998163e+00]
 [ 0.00000000e+00  0.00000000e+00  0.00000000e+00  1.00000000e+00]]

This is how I am defining my chain.
```
my_chain = Chain(name='robot_arm', links=[
    OriginLink(),
    URDFLink(
      name="link1",
      origin_translation =[0, 0, 0],
      origin_orientation =[0, 0, 0],
      rotation=[0, 0, 1], 
    ),
    URDFLink(
      name="link2",
      origin_translation =[0, 0, 5],
      origin_orientation =[0, 0, 0],
      rotation=[0, 1, 0],
    ),
    URDFLink(
      name="link3",
      origin_translation =[0, 0, 5],
      origin_orientation =[0, 0, 0],
      rotation=[0, 1, 0],
    ),
 ])
```

and this is how I am calling the inverse kinematics
`my_chain.inverse_kinematics([x, y, z])`

Phylliade, 2022-04-04:
Hello @nickredsox,

Thanks for your detailed issue;

Whereas it's possible to create URDF links by hand, it's recommended to use the URDF file of your robot to create your chain.

[This file](https://github.com/Interbotix/interbotix_ros_manipulators/blob/main/interbotix_ros_xsarms/interbotix_xsarm_descriptions/urdf/px100.urdf.xacro) looks like a valid URDF file your robot: what do you think?

nickredsox, 2022-04-09:
Thanks for the response and the link. I will have to look at that link in more detail later because I was hoping to avoid the URDF files. It has been a long time since I used ROS and am not particularly familiar with it

closed 2022-06-20 by stale[bot]