Phylliade/ikpy #129: Supporting ik interpolation between two positions

DoWhileGeek, 2022-11-29:
First off, this project is fantastic, it seriously helps with DIY arm development! I'm mostly asking if this is something within the purview of this library, but how do y'all feel about supporting interpolated values between two points, your current, and target position? I'm working on a pick and place (of a sort) robot, and a large part of the job is picking up items from one or several places, and feeding it fairly accurately into a deep container. I know fora fact when I first got my homemade ik model off the ground, the arm would like, "swing" into position, and the only way around it was instead of one movement, many shorter segments of movements that shortened the amount the arm could "swing" (I'm sure theres a technical term I'm butchering, sorry).

I'm imagining an option in `inverse_kinematics()` that takes some sort of a parameter for tokenizing the movement into a list of movements. Maybe the value is a numSegments integer, or the default length unit, like millimeters per segment. Taking the concept to the nth degree so it doesnt feel like for loop syntactic sugar, I'm envisioning that target could be a list of values, so the model could be animated at a higher level, maybe even supporting velocity ~~suggestions~~ calculations given a max velocity value, maybe even per axis?

If thats outside of this libraries mission, I totally get it, just wanted to know if you guys were open to those sort of features as pull requests.

Phylliade, 2022-12-02:
Hello @DoWhileGeek !

Actually that's a very interesting feature, and certainly to be added in IKPy!

A few quick thoughts:

* I'm pretty sure i've written a simple snippet in the past, that splits a trajectory into multiple target positions. However, i'm unable to find it after a quick lookup :shrug: 
* IMO the main difficulty of this feature is a design constraint: To be more precise, we want to add the notion of **trajectory** into IKPy, which can immediately translates into a sequential list of target positions, but can also be a more abstract object (i.e. a continuous trajectory is a curve, that can also includes velocity information)
* Once we have a curve, it's easy to sample from it to get the list of targets; 
* So the main issue is "How to define a curve in a simple ikpyesque ( ;) ) way?": There are many ways to define curves, but maybe choosing the right one is not so easy
* For example, with your picking robot, you want your robot to follow a certain trajectory (i.e avoiding colliding your walls), instead of going in an undefined line between your starting position and your target

What are your thoughts on this? 

Especially, how would you define your trajectory in your use case?

DoWhileGeek, 2022-12-10:
Sorry, accidental click on the close button. Let me think on this a bit before I respond, but those are super valid points.

Itssshikhar, 2024-08-01:
Hey guys! I am doing somewhat similar thing for my own Robot, trying to do pick and place. This discussion looks promising.
Did you guys reached some sort of conclusion for this?

Phylliade, 2024-08-31:
Hello @Itssshikhar, this is still in the TODO state!

Phylliade, 2026-10-27:
Shipped in 4.3.0: `chain.inverse_kinematics_trajectory_frame(frames, initial_position=current_joints)` solves a sequence of frames, each step starting from the previous solution, so the arm follows one branch of solutions instead of swinging to another one, and returns a `(n_steps, n_links)` array. `ikpy.utils.geometry.interpolate_frames(start, end, n_steps)` builds the frames between two poses, straight line in position and constant-speed rotation in orientation. See the "Moving along a trajectory" section of the README and the release notes: https://github.com/Phylliade/ikpy/releases/tag/v4.3.0

closed 2026-10-27 by Phylliade