Phylliade/ikpy #130: How to add constrains to the chain to ensure the robot arm is always above the workbench?

FredTrumpSenior, 2023-01-12:
How can I add constrains to ensure that the arm is always above the workbench without colliding with it. 
[image]
ithout colliding with it.

Phylliade, 2023-04-10:
Hello @FredTrumpSenior,

For the moment, there is no direct way to add constraints on links positions. 

However you can add a constraint in joint angles by defining joint bounds:
For example you can try to [add a bound](https://github.com/Phylliade/ikpy/blob/a32b3f3d3a02858b8e727941782a703e14e291b0/src/ikpy/link.py#L83) for the 3rd joint (the joint before the link that goes into the workbench) to be only positive, this way the link will never be in the negative Zs:

FredTrumpSenior, 2024-03-07:
Thanks for your kind reply

closed 2024-03-07 by FredTrumpSenior