Phylliade/ikpy #132: The ValueError：“x0” is infeasible

cidxb, 2023-01-31:
Hi.
Following the IKpy Quickstart.
I found that the inverse_kinematics function is broken.
By the Error info, it should be the  inverse_kinematics. py line 135, 
the res=scipy.optimize.least_squares went wrong. Something about the bounds.
The ValueError： “x0” is infeasible

Is that because of my UDRF mode problem?
Anyone can help me to fix it ?
Thx

cidxb, 2023-02-01:
Solved:
Basically it is a problem of the bound range of joint.

closed 2023-02-01 by cidxb

cidxb, 2023-03-31:
> Same error, how did you fix it?

Check for your XML model on its joint's range, this problem happens, because the action is out of the limit of the joint range.

irmakguzey, 2023-10-05:
Hello,
Sorry to open this again but I have been struggling with this issue for a while as well. But wouldn't the better implementation of this be to return the most possible joint values? Rather than giving an error?

cidxb, 2023-10-10:
> Hello, Sorry to open this again but I have been struggling with this issue for a while as well. But wouldn't the better implementation of this be to return the most possible joint values? Rather than giving an error?

The problem is ikpy is trying to solve a reverse kinetic problem with given conditions, and the joint values boundary is the conditions you give, and then conduct a iterative solving process. So I think it is very hard to "return the most possible joint values", because the  joint values boundary can be arbitrary based on the actually limitation of your research, and also (I think) related to the actual 3D model...

mrudorfer, 2023-12-07:
I stumbled over this as well, and for me it was the `initial_position` given to the IK solver that was in conflict with the joint limits - it makes sense that this raises an error. Perhaps the error message could be more specific though.

Note that it is not necessarily obvious, I was using a UR5e robotic arm with 3-finger gripper, but I didn't really care about the gripper joints yet, so I used this as initial joint pos:
```
initial_position = [0] + [m.getPositionSensor().getValue() for m in ur_motors] + [0, 0, 0, 0]
```
However, it turns out the finger joints have limits that do not include zero, so I needed to change it to e.g.:
```
initial_position = [0] + [m.getPositionSensor().getValue() for m in ur_motors] + [0, 1, 1, -1]
```