Phylliade/ikpy #133: How to use ikpy together with any other path planning algorithm to achieve obstacles avoidance?

johnlim00, 2023-02-03:
Currently working on a project of simulating autonomous robotic arms. I wonder is it possible to do it by using ikpy in my obstacles avoidance algorithm. Im able to use ikpy to move my ur5e robotic arms to the target object, but i wanted to make the arms automatically avoid any obstacles (if there's any) and reach the target object. Still stucked at the obstacles avoidance part, is there any tips or solution for this?

gurselturkeri, 2023-02-26:
Hi, how did you do set goal and arms moving to goal? Also for obstacles avoidance maybe you can use lidar sensor.

Phylliade, 2023-04-10:
Hello @johnlim00 !

IKPy's general philosophy is to be a low-level ik library, that can be used in higher levels trajectory/path planning algorithms, so your point makes sense!

The general approach is to cut your trajectory into multiple keypoints, in order to avoid the obstacles, and then use IKPy do the IK for you between the different keypoints

See more info here:
#129

shaneraphel, 2026-09-18:
Owner note on keypoints is the right split: IK between named poses, a separate field for obstacles.

Reachlock keeps that split. The arm kernel refuses an unreachable target. The clearance kernel is integer `r²` on disks, so the same pose and disks hash to the same digest. Empty field is absence.

https://github.com/shaneraphel/aletheia-reachlock

This is a related playable artifact, not a planner for IKPy.

Phylliade, 2026-10-27:
The split described above is now supported directly: compute the waypoints that avoid your obstacles with the planner of your choice, and give them as frames to `chain.inverse_kinematics_trajectory_frame(frames, initial_position=current_joints)` (4.3.0). Each waypoint is solved from the solution of the previous one, so the joint motion stays continuous along the path. Obstacle avoidance itself stays out of IKPy, which remains the low-level IK layer. See the "Moving along a trajectory" section of the README and the release notes: https://github.com/Phylliade/ikpy/releases/tag/v4.3.0

closed 2026-10-27 by Phylliade