Phylliade/ikpy #134: ValueError: Unknown joint type: continuous

gurselturkeri, 2023-02-26:
Hi, I get an error when I run my python script.

```
Traceback (most recent call last):
  File "/home/gursel/robotics_ws/src/robotic_arm/2_inverse_kinematics_solution.py", line 72, in <module>
    main()
  File "/home/gursel/robotics_ws/src/robotic_arm/2_inverse_kinematics_solution.py", line 64, in main
    joint_trajectory_object = Trajectory_publisher()
  File "/home/gursel/robotics_ws/src/robotic_arm/2_inverse_kinematics_solution.py", line 28, in __init__
    self.robot_initialize(urdf_file)
  File "/home/gursel/robotics_ws/src/robotic_arm/2_inverse_kinematics_solution.py", line 43, in robot_initialize
    self.kuka_robot = ikpy.chain.Chain.from_urdf_file(urdf_file)
  File "/home/gursel/.local/lib/python3.10/site-packages/ikpy/chain.py", line 324, in from_urdf_file
    links = URDF.get_urdf_parameters(urdf_file, base_elements=base_elements, last_link_vector=last_link_vector, base_element_type=base_element_type, symbolic=symbolic)
  File "/home/gursel/.local/lib/python3.10/site-packages/ikpy/urdf/URDF.py", line 250, in get_urdf_parameters
    raise ValueError("Unknown joint type: {}".format(joint_type))
ValueError: Unknown joint type: continuous
```

Phylliade, 2023-04-10:
Hello @gurselturkeri ,

Can you paste your URDF file?

StrangerChamber, 2023-04-25:
I'm having the same issue. 

here is my urdf file:

<?xml version="1.0"?>
<robot name="TIAGo Steel" xmlns:xacro="http://ros.org/wiki/xacro">
  <link name="base_link">
    <visual>
      <origin xyz="0 0 0.06" rpy="3.1407962 0 0"/>
      <geometry>
        <cylinder radius="0.265" length="0.276"/>
      </geometry>
    </visual>
    <collision>
      <origin xyz="0 0 0.06" rpy="3.1407962 0 0"/>
      <geometry>
        <cylinder radius="0.265" length="0.276"/>
      </geometry>
    </collision>
    <visual>
      <origin xyz="-0.201 0.1062 0.28" rpy="3.1407962 0 0"/>
      <geometry>
        <cylinder radius="0.007" length="0.173"/>
      </geometry>
    </visual>
    <collision>
      <origin xyz="-0.201 0.1062 0.28" rpy="3.1407962 0 0"/>
      <geometry>
        <cylinder radius="0.007" length="0.173"/>
      </geometry>
    </collision>
    <visual>
      <origin xyz="-0.201 -0.1062 0.28" rpy="3.1407962 0 0"/>
      <geometry>
        <cylinder radius="0.007" length="0.173"/>
      </geometry>
    </visual>
    <collision>
      <origin xyz="-0.201 -0.1062 0.28" rpy="3.1407962 0 0"/>
      <geometry>
        <cylinder radius="0.007" length="0.173"/>
      </geometry>
    </collision>
  </link>
  <joint name="hingejoint_12445" type="continuous">
    <parent link="base_link"/>
    <child link="caster_back_left_1_link"/>
    <axis xyz="0 0 1"/>
    <origin xyz="-0.1735 0.102 -0.0335" rpy="0 0 0"/>
  </joint>
  <link name="caster_back_left_1_link">
    <visual>
      <origin xyz="-0.003 0 -0.022" rpy="0 0 0"/>
      <geometry>
        <box size="0.036 0.032 0.04"/>
      </geometry>
    </visual>
    <collision>
      <origin xyz="-0.003 0 -0.022" rpy="0 0 0"/>
      <geometry>
        <box size="0.036 0.032 0.04"/>
      </geometry>
    </collision>
  </link>
  <joint name="hingejoint_12448" type="continuous">
    <parent link="caster_back_left_1_link"/>
    <child link="caster_back_left_2_link"/>
    <axis xyz="0 -3.6732051e-06 1"/>
    <origin xyz="-0.016 0 -0.04" rpy="-1.5708 0 0"/>
  </joint>
[... 1016 lines cut ...]
    <child link="cap screw(3)"/>
    <origin xyz="0.0477 -0.008 -0.075" rpy="3.1415927 -1.5707927 3.1415927"/>
  </joint>
  <link name="cap screw(2)">
  </link>
  <joint name="front_cap screw(2)_joint" type="fixed">
    <parent link="front"/>
    <child link="cap screw(2)"/>
    <origin xyz="0.0477 0.008 -0.075" rpy="3.1415927 -1.5707927 3.1415927"/>
  </joint>
  <link name="cap screw(1)">
  </link>
  <joint name="front_cap screw(1)_joint" type="fixed">
    <parent link="front"/>
    <child link="cap screw(1)"/>
    <origin xyz="0.0477 0.008 -0.043" rpy="3.1415927 -1.5707927 3.1415927"/>
  </joint>
  <link name="cap screw">
  </link>
  <joint name="front_cap screw_joint" type="fixed">
    <parent link="front"/>
    <child link="cap screw"/>
    <origin xyz="0.0477 0.008 -0.0108" rpy="3.1415927 -1.5707927 3.1415927"/>
  </joint>
</robot>

gurselturkeri, 2023-04-29:
> I'm having the same issue.
>
> here is my urdf file:
>
>
> <?xml version="1.0"?>
> <robot name="TIAGo Steel" xmlns:xacro="http://ros.org/wiki/xacro">
>   <link name="base_link">
>     <visual>
>       <origin [...]

In ros2 humble version it works clearly.

gurselturkeri, 2023-04-29:
@StrangerChamber In ros2 humble version it works clearly.

EdgarJRobles, 2024-05-21:
I faced the same issue running a urdf  from ROS here: [https://github.com/ros-industrial/abb/tree/kinetic-devel](url)  on webots.
webots implement ikpy library to calculate de inverse kinematics.

Heavy industrial robots usually a piston to support 2 axis  heavy loads an that why it have invinite revolute ROS called "continuous" joint type.

It looks like there is not definition for continuous joint type on URDF.py
`       joint_type = joint.attrib["type"]
        if joint_type not in ["revolute", "prismatic", "fixed"]:
            raise ValueError("Unknown joint type: {}".format(joint_type))

        axis = joint.find("axis")
        if axis is not None:
            if joint_type == "revolute":
                rotation = [float(x) for x in axis.attrib["xyz"].split()]
                translation = None
            elif joint_type == "prismatic":
                rotation = None
                translation = [float(x) for x in axis.attrib["xyz"].split()]
            elif joint_type == "fixed":
                warnings.warn(
                    "Joint {} is of type: fixed, but has an 'axis' attribute defined. This is not in the URDF spec and thus this axis is ignored".format(
                        joint.attrib["name"]
                    )
                )
            else:
                raise ValueError(
                    "Unknown joint type with an axis: {}, {}".format(joint_type, axis)
                )
`

mhortman, 2024-06-26:
I think you can replace convolute with revolute, and add some limits on to the revolute, and it should work.  I am testing this theory now.

MichaelrMentele, 2024-08-07:
Having the same issue -- any update on workaround @mhortman ?

mhortman, 2024-08-21:
well doing that got it to parse, but my urdf file doesn't seem to send my arm to the right xyz coordinates, so not 100% sure.

Phylliade, 2024-09-01:
Hello everyone,

I will add this feature asap, in the stylé of @mhortman !

wzds2015, 2024-11-21:
@Phylliade what is the idea to add support for continuous joint? I changed continuous to convolute to make ik work. But I think they are not the same mechanism. Still want to switch to continuous joint.

closed 2026-10-26 by Phylliade