Phylliade/ikpy #138: Problème pour atteintre la position ET l'orientation 3 axes souhaitée

Benjamin797, 2023-04-10:
Bonjour,

Cela fait plus d'un mois que j'explore votre algorithme et mon but est de résoudre de l'inverse kinematic pour des bras articulés que je créer directement en python avec vos fonctions. Mon projet me contraint à résoudre le problème en position ET en orientation sur les 3 axes (orientation_mode = 'all'). Cependant, malgré plusieurs tentatives, l'optimisation me sort des résultats où la position est atteinte mais pas l'orientation. est ce que votre algorithme fonctionne bien pour une résolution orientation_mode ) 'all'? si oui, pouvez vous m'aider à résoudre ce problème? J'en aurait réellement besoin. 
Voici mon bout de code permettant la résolution de l'ik. Elle est faite en deux temps mais il n'arrive pas a résoudre l'orientation 3 axes alors que je sais que c'est possible de l'atteindre (ce que j'ai réussi à prouvé antérieurement dans mon projet).

Merci d'avance.

`
  res = chain.inverse_kinematics(target.coordToList())
  res = chain.inverse_kinematics(target_position=target.coordToList(), target_orientation=targ_ori, 
                                 orientation_mode= 'all', initial_position = res)
  
  
  position = chain.forward_kinematics(res)[:3, 3]
  orientation = chain.forward_kinematics(res)[:3, :3]
  
  np.testing.assert_almost_equal(orientation, targ_ori, decimal=5)
  np.testing.assert_almost_equal(position, target.coordToList(), decimal=3)
`

Benjamin797, 2023-05-03:
One solution I found is the randomly change the inital position for each joint and iterate the resolution until the position and orientation is reached (with a cost fonction that compare the real position/orientation computed with the exepected one). Althought this method takes a really long time (1 minute sometimes) to reach the target wanted. Normally, if Ikpy workd for orientation_mode = all, i shoudln't iterate like that, one execution of the inverse_kinematic function should be enough. Is this orientation_mode really works?
Thank you

niklasconbotics, 2023-06-09:
Hi Benjamin,
I have the same problem! can you provide the code? Or did you found a more convinced solution?

Phylliade, 2026-10-26:
`orientation_mode="all"` does work, but it is a local solver: it converges to the nearest solution of the starting point, and on a 6-DOF arm a position-only pre-solve (your first step) often lands in a configuration from which the full orientation is not reachable without passing through a joint limit. Checking on a Niryo One with IKPy 4.2.0, the full-frame IK from the default initial position reaches a frame taken from a forward kinematics within 3e-8 in position and 3e-9 in orientation.

Two things that usually fix the cases you describe:

- Solve in one step with `orientation_mode="all"`, starting from your current joint configuration (`initial_position`), rather than from a position-only solve.
- When the target is far from the current pose, try a few starting points spread inside the joint limits and keep the best; your random restarts did exactly that. A built-in multi-start option is tracked in #144.

Without your chain and target, I cannot check for a bug on your side. Closing for now; please reopen with the links of your chain (the output of `print(chain)` and the link parameters) and a target that you know is reachable, and I will look at it.

closed 2026-10-26 by Phylliade