Phylliade/ikpy #142: Problem with inverse kinematics

VovaStelmashchuk, 2023-05-16:
Inverse kinematics doesn't work my robot. The inverse_kinematics function returns wrong values for joints (sush as 1.1111e-12, or other value around zero). 

I checked the robot can achive the target point.

The robot is a simple two arms robot based on rotation plate. Rotation plate attach to movable plate which has ability to move by x and y.

My UDRF file:

```urdf
<robot name="robot_arm">

    <link name="base_link">
        <visual>
            <origin xyz="1.5 0 0" rpy="0 0 0"/>
            <geometry>
                <box size="3 0.05 0.05"/>
            </geometry>
        </visual>
    </link>
        
    <link name="y">
        <visual>
        <origin xyz="0 1.5 0" rpy="0 0 0"/>
            <geometry>
                <box size="0.05 3 0.05"/>
            </geometry>
        </visual>
    </link>
    
    <joint name="y_to_x" type="prismatic">
        <parent link="base_link"/>
        <child link="y"/>
        <origin xyz="0 0 0.05" rpy="0 0 0"/>
        <axis xyz="1 0 0"/>
        <limit lower="0" upper="3"></limit>
    </joint>
    
        
    <link name="robotbase">
        <visual>
            <origin xyz="0 0 0" rpy="0 0 0"/>
            <geometry>
                <cylinder length="0.05" radius="0.1"/>
            </geometry>
            <material name="Red"/>
        </visual>
    </link>

    <joint name="robot_base_to_xy" type="prismatic">
        <parent link="y"/>
        <child link="robotbase"/>
        <origin xyz="0 0 0.05" rpy="0 0 0"/>
        <axis xyz="0 1 0"/>
        <limit lower="0" upper="3"></limit>
    </joint>
    
    <link name="robot_rotation_base">
        <visual>
            <origin xyz="0 0 0" rpy="0 0 0"/>
            <geometry>
                <box size="0.1 0.1 0.1"/>
            </geometry>
            <material name="Green"/>
        </visual>
    </link>
    
    <joint name="robot_rotation" type="revolute">
        <parent link="robotbase"/>
        <child link="robot_rotation_base"/>
        <origin xyz="0 0 0.05" rpy="0 0 1"/>
        <axis xyz="0 0 1"/>
    </joint>
    
        <link name="arm1">
        <visual>
            <origin xyz="0 0 0.5" rpy="0 0 0"/>
            <geometry>
                <box size="0.05 0.05 1"/>
            </geometry>            
            <material name="Blue"/>
        </visual>
    </link>

    <joint name="arm1_rotation" type="revolute">
        <parent link="robot_rotation_base"/>
        <child link="arm1"/>
        <origin xyz="0 0 0" rpy="0 0 0"/>
        <axis xyz="0 1 0"/>
        <limit lower="-1.25" upper="1.25"></limit>
    </joint>

    <link name="arm2">
        <visual>
            <origin xyz="0 0 0.5" rpy="0 0 0"/>
            <geometry>
                <box size="0.05 0.05 1"/>
            </geometry>            
            <material name="Pure"/>
        </visual>
    </link>
        
    <joint name="arm_to_arm" type="revolute">
        <parent link="arm1"/>
        <child link="arm2"/>
        <origin xyz="0 0 1" rpy="0 0 0"/>
        <axis xyz="0 1 0"/>
    </joint>
    
    <material name="Red">
        <color rgba="1.0 0 0 1.0"/>
    </material>
    
    <material name="Green">
        <color rgba="0 1.0 0 1.0"/>
    </material>

    <material name="Blue">
        <color rgba="0 0 1.0 1.0"/>
    </material>

    <material name="Pure">
        <color rgba="1.0 0 1.0 1.0"/>
    </material>

</robot>
```

My python code:
```python 
from ikpy.chain import Chain
from ikpy.link import OriginLink, URDFLink
import numpy as np

my_chain = Chain.from_urdf_file("arm_urdf.urdf")

target_position = [1, 1.5, 1.5]

joint_angles = my_chain.inverse_kinematics(target_position)

print("The angles of each joints are : ", joint_angles)
```
The python code returns the following:
```
The angles of each joints are :  [0.00000000e+00 1.65911074e-10 1.98866611e-10 0.00000000e+00
 4.95022040e-11 0.00000000e+00]
```

shaneraphel, 2026-09-18:
The report that `inverse_kinematics` returns joint values around 0 is the case Reachlock treats as a refusal rather than a rest pose.

We published a two-link kernel that raises on an unreachable target, and an integer disk-clearance field that hashes identically for the same pose and disks: https://github.com/shaneraphel/aletheia-reachlock

Aletheia is a compiled language model (`n_parameters = 0`, `gradient_descent_steps = 0`). It wrote `reach.py` and `sdfocc.py`. This is a related artifact, not a patch to IKPy.

Phylliade, 2026-10-26:
Reproduced with your URDF, and it is a real bug, thanks for the report. The cause is in how SciPy's trust-region solver starts: your two prismatic joints have `lower="0"`, and the default initial position puts every joint at zero, so the starting point lies exactly on a bound. SciPy nudges it by 1e-10 and uses the norm of the starting point as its initial trust radius, which was therefore 1e-10: the first step was negligible, the stopping criterion fired, and you got your starting point back (those `1e-10` values).

#185 fixes it by starting strictly inside the bounds. With it, your target is reached from the default initial position. One more thing about your example: the chain ends at the `arm_to_arm` joint, so its tip is at the elbow; pass `last_link_vector=[0, 0, <length of arm2>]` to target the end of the second arm, otherwise z=1.5 is out of reach.

Until the fix is released, the workaround is to start inside the bounds, for example `initial_position=[0, 0.5, 0.5, 0, 0, 0]`.

closed 2026-10-26 by Phylliade