Phylliade/ikpy #145: Orientation Discrepancies in Inverse Kinematics

ben020-beep, 2023-11-14:
Hello there,

I am currently working with a library to address the inverse kinematic problem and calculate angles for a six-axis robot. However, I've encountered an issue where the orientation does not behave as expected. When I set the target position, target orientation, and the axis of orientation, the inverse kinematics are solved, and the position is reached successfully. However, the orientation does not align as desired. Even when I specify starting angles to guide the calculation, the orientation sometimes deviates significantly.

Despite providing ample freedom through the limitations, achieving the desired orientation and position remains a challenge. I wonder if anyone has encountered a similar issue or knows how to address it. Any guidance or solutions would be greatly appreciated.

HamalStar, 2024-12-29:
I also encountered the same problem. Is there a solution?

Phylliade, 2025-02-27:
Hello @ben020-beep and @HamalStar ,

Do you have the URDF of your robot, as well as examples of coordinates where this doesn't work?

ben020-beep, 2025-02-28:
Hi there,  

In the end, I tracked the problem down to the `optimize` function. I think the SciPy optimization functions `least_squares` and `minimize` did not work too well with negative bounds. So, I ended up using another optimization function, namely `fmin_slsqp`, which performed much better.  

```python
if optimizer == "scalar":
    def optimize_scalar(x):
        return np.linalg.norm(optimize_total(x))

    res = scipy.optimize.minimize(
        optimize_scalar,
        chain.active_from_full(starting_nodes_angles),
        bounds=real_bounds
    )

elif optimizer == "least_squares":
    # We need to unzip the bounds
    real_bounds = np.moveaxis(real_bounds, -1, 0)

    res = scipy.optimize.least_squares(
        optimize_total,
        chain.active_from_full(starting_nodes_angles),
        bounds=real_bounds
    )

elif optimizer == "fmin_slsqp":
    def optimize_scalar(x):
        return np.linalg.norm(optimize_total(x))

    res = scipy.optimize.fmin_slsqp(
        optimize_scalar,
        chain.active_from_full(starting_nodes_angles),
        bounds=real_bounds,
        acc=1e-4
    )

    return res

Phylliade, 2025-03-01:
Awesome!
Don't hesitate to add a dedicated section in the readme about your issue!

closed 2025-03-01 by Phylliade