Phylliade/ikpy #158: IKPY Smooth Motion: Inverse Kinematics using Quintic Trajectory

GaboBoho, 2024-09-26:
I'm a student part of a team working on a 3D Printed Humanoid based of the Poppy Open Source Project. We are looking into incorporating the IKPY library for both inverse kinematics of the arms and feet. We have created a program that calculates the quintic trajectory with each way point we calculate the inverse kinematics using IKPY to determine the angles for the motors. An issue we are running into is that for each inverse kinematics calculation it does not take the previous into account meaning that it might create erratic movement. Is there any way we could change the library so that it takes the previous angle into account and movement becomes smooth?

Phylliade, 2024-10-05:
Hello @GaboBoho, 

Actually when you give the starting joint angles as parameter, it defines both position and orientation.

Can you provide a recording of the erratic movement of your robot?

Phylliade, 2026-10-27:
Shipped in 4.3.0: give your quintic trajectory as a list of frames to `chain.inverse_kinematics_trajectory_frame(frames, initial_position=current_joints)`, or its positions and orientations to `inverse_kinematics_trajectory`. Each step starts from the solution of the previous one, which is what keeps the motion smooth, and `regularization_parameter` pulls each step further towards the previous one if needed. See the "Moving along a trajectory" section of the README and the release notes: https://github.com/Phylliade/ikpy/releases/tag/v4.3.0

closed 2026-10-27 by Phylliade