Phylliade/ikpy #159: Calculate IK based on the world frame

joel0115, 2024-11-17:
Hi, thanks for your impressive works, sorry for my disturbing.
I used the following code to test the IK calculator.
However, when I plot the chain, I found that the chain's base is not located at the world's origin (0,0,0).
And I realized that IK is calculated based on the chain base's frame. But I want to calculate IK based on world frame.
So, is it possible to do one of the following or both?

1. Locate my chain at the world's origin.
2. Calculate IK based on the world's frame. i.e. [0.6042, 0.1777, 0.3502] means the coordinate w.r.t. world frame, instead of the chain base frame.

Thanks in advance!

```
my_chain = ikpy.chain.Chain.from_urdf_file(urdf_path)

target_position = [0.6042, 0.1777, 0.3502]
print("The angles of each joints are : ", my_chain.inverse_kinematics(target_position))

real_frame = my_chain.forward_kinematics(my_chain.inverse_kinematics(target_position))
print("Computed position vector : %s, original position vector : %s" % (real_frame[:3, 3], target_position))

fig, ax = plot_utils.init_3d_figure()
my_chain.plot(my_chain.inverse_kinematics(target_position), ax, target=target_position)
plt.xlim(-1 ,1)
plt.ylim(-1, 1)
plt.savefig("IK.png")
```
[image]

joel0115, 2024-11-17:
An additional problem is that, my chain consists of arm and finger parts, but the IK "seems" only considering the arm's end, not the finger's tool center pose (tcp).
Is it due to my urdf file or do I need to pass some parameters to let the chain realize it?
Thanks in advance!

Phylliade, 2025-02-27:
Hello @joel0115,

Regarding your first problem: I'm not sure to understand: the base frame is by definition at the world origin.

Regarding your second problem: can you provide the URDF and code you're using?

Phylliade, 2026-10-26:
To answer the original question: the base of a chain *is* the world origin for IKPy. `forward_kinematics` and `inverse_kinematics` work in the frame of the first link of the chain, so a target given to `inverse_kinematics` is expressed in that frame. If your URDF places the robot somewhere else in the world (for example with a fixed joint between `world` and `base_link`), start the chain at the world link with `base_elements=["world"]` so that the fixed offset is part of the chain, and targets can then be given in world coordinates. If the offset is not in the URDF, transform your target into the base frame before calling the IK.

For the second question (tool center point of a gripper): the chain ends where its last link ends. Either include the finger joints in `base_elements`, or give the TCP offset with `last_link_vector`.

Closing, as there is no change needed in the library; feel free to open a new issue with your URDF if one of these does not work.

closed 2026-10-26 by Phylliade